tttapa/Control-Surface #301: Cubase Integration

masydoblig, 2020-10-16:
Hi Peter

Haven't been here in a while in while :)

Your project has come along way and looks amazing....well done 
i have finally managed to get my EQ module working using your code. the problem i had was the Generic MIDI controller editor in Cubase. 
There is very little documentation on it and Steinberg don't seem to have a clue how it works......hahaha (and they made it)

So essentially in the generic editor any value/command that Cubase has can be received or transmitted 
but the drop down lists to select them are very incomplete.
so i have had to resort to opening the config files in an editor and manually entering the functions i want to control.

For instance the mixer input filters are not in the drop down list. 
so to use them as a value/action i had to guess what the descriptor in the config file was. (took ages to work out what it was some use capitals some don't) it was like cracking a password. **InputFilter:param3**

[image]

extract from exported Cubase generic config file

[image]

So now i have a fully functional Cubase EQ strip......WIN

[image]

Do you have any tips on how to go into the Cubase program files and extract the midi function list 
I would like to start a Git page for the generic editor since Steinberg wont help

PS have you made any progress with the motor fader support?

i found this guy on you tube and had a look at his Git page seems to be a good thing.
https://github.com/saarts/MIDI/blob/master/MIDI_flying_fader/MIDI_flying_fader.ino

Cheer's Sam

tttapa, 2020-10-17:
I'm afraid I can't really help you with that, I've never used Cubase, and I'm not familiar with reverse engineering desktop software.

I haven't had time to work on motor fader support, my studies have priority right now, and while writing code in my free time is not really an issue, I didn't get around building the hardware setup to continue working on the motor faders.

oscarosky, 2020-11-21:
Puedes utilizar la implementacion MIDI del controlador  steinberg CC121 
[https://www.steinberg.net/forums/viewtopic.php?t=58818](url)