tttapa/Control-Surface #302: oled library

masydoblig, 2020-10-17:
Hey Peter

How hard is it to use a different display?

i have an sh1106 SPI 128x64 which uses the U8g library?

Cheers Sam

tttapa, 2020-10-17:
Yes, you can implement the `DisplayInterface` class: https://github.com/tttapa/Control-Surface/blob/master/src/Display/DisplayInterface.hpp

masydoblig, 2020-10-18:
Hi Peter
Trying to get this to work with the DisplayInterface  class and im clearly not getting it right 
if you get a chance can you have a look.

using the <Adafruit_SH1106.h> library which works with the display (tested with examples provide in library)
will be testing it on my Due board connected to the SPI headers 

/**
 * An example demonstrating the use of DisplayElement%s to display information
 * from the DAW on a small OLED display.
 *
 * @boards  Mega
 * 
 * Connections
 * -----------
 * 
 * - 5:  Push button (to ground)
 * - 6:  Push button (to ground)
 * - 51  OLED Data/D1 (SPI MOSI)
 * - 52: OLED Clock/D0 (SPI SCK)
 * - 49: OLED Data/Command
 * - 53: OLED Cable Select
 * 
 * Add a capacitor between the reset pin of the display and ground, and a 
 * resistor from reset to 3.3V. The values are not critical, 0.1µF and 10kΩ 
 * work fine.  
 * You do need some way to reset the display, without it, it won't work.  
 * Alternatively, you could use an IO pin from the Arduino to reset the 
 * display, but this just "wastes" a pin.
 * 
 * Behavior
 * --------
 * 
 * - The time (bars, beats, fraction), play and record status are shown at the 
 *   top of the display.
 * - For each of the 8 first tracks, a VU level meter with peak indicator and
 *   a V-Pot ring showing the pan are displayed, as well as the the mute, solo 
 *   and record arm status.
 * - Two tracks are displayed at once. By pressing the push buttons connected
 *   to pins 5 and 6, you can cycle through four banks to display all 8 tracks.
 * 
 * Mapping
 * -------
 * 
 * Map "Control Surface" as a Mackie Control Universal unit in your DAW.
 * 
 * @note    There seem to be some differences in the way some applications 
 *          handle VU meters: some expect the hardware to decay automatically,
 *          some don't.  
 *          If you notice that the meters behave strangely, try both 
 *          MCU::VUDecay::Hold and MCU::VUDecay::Default, or try a different 
 *          decay time.
 * 
 * Written by PieterP, 2019-11-12  
 * https://github.com/tttapa/Control-Surface
 */

#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
// Include the display interface you'd like to use
#include <Display/DisplayInterface.hpp>
#include <Adafruit_SH1106.h>

// ----------------------------- MIDI Interface ----------------------------- //
// ========================================================================== //

/*
   Instantiate a MIDI interface to use for the Control Surface.
*/
[... 190 lines cut ...]
// Bank seting
BankDisplay bankDisp[] = {
  // first track of the bank (1), position (0, 50), font size (2)
  {display, bank, 1, {0, 50}, 2, WHITE},
  {display, bank, 2, {64, 50}, 2, WHITE},
};

// --------------------------------- Setup ---------------------------------- //
// ========================================================================== //

void setup() {
  // Correct relative mode for MCU rotary encoders

  RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin(); // Initialize Control Surface
 
}

// ---------------------------------- Loop ---------------------------------- //
// ========================================================================== //

void loop() {
  Control_Surface.loop();  // Refresh all elements

}

tttapa, 2020-10-19:
`DisplayInterface` is an abstract interface. It declares a set of abstract (pure virtual) drawing functions for displays. Control Surface uses these functions to draw to the displays, but Control Surface doesn't know how these functions are implemented, the implementation depends on the display library you're using, and it's up to the user to create an adapter between Control Surface and the display library by implementing the pure virtual functions of `DisplayInterface`.

As an example, you could have a look at the [`DisplayInterfaceSSD1306`](https://github.com/tttapa/Control-Surface/blob/21a09383bb3eaa5f1835184a63fdcdb4b8f5799b/src/Display/DisplayInterfaces/DisplayInterfaceSSD1306.hpp), it should be almost identical for the Adafruit_SH1106 library.

masydoblig, 2020-11-01:
Hi Peter 

Can you have a look at the below i cannot get the display interface working at all 
It is for the Adafruit_ST7735 screen library 
i have been trying to make a new adapter to the display interface and its not working at all 

#include <Encoder.h>
#include <Control_Surface.h>
#include <Display/DisplayInterfaces/DisplayInterfaceST7735.hpp>

 // ----------------------------- MIDI Interface ----------------------------- //
 // ========================================================================== //

 /*
    Instantiate a MIDI interface to use for the Control Surface.
 */

    USBMIDI_Interface midi;
// USBDebugMIDI_Interface midi(115200);

// ----------------------------- Display setup ------------------------------ //
// ========================================================================== //

//constexpr uint8_t SCREEN_WIDTH = 128;
//constexpr uint8_t SCREEN_HEIGHT = 64;

constexpr int8_t TFT_DC = 8;    // Data/Command pin of the display
constexpr int8_t TFT_RST = -1; // Use the external RC circuit for reset
constexpr int8_t TFT_CS = 9;    // Chip Select pin of the display

//constexpr uint32_t SPI_Frequency = SPI_MAX_SPEED;

// Instantiate the displays
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_RST);
float p = 3.1415926;

// --------------------------- Display interface ---------------------------- //
// ========================================================================== //

// Implement the display interface, specifically, the begin and drawBackground
// methods.
class MyST7735_DisplayInterface : public ST7735_DisplayInterface {
public:
    MyST7735_DisplayInterface(Adafruit_ST7735& display)
        : ST7735_DisplayInterface(display) {}

    void begin() override {
        // Initialize the Adafruit_SSD1306 display
        if (!tft.begin())
            FATAL_ERROR(F("ST7735 allocation failed."), 0x1306);

        // If you override the begin method, remember to call the super class method
        ST7735_DisplayInterface::begin();
    }

    void drawBackground() override { tft.drawLine(1, 8, 126, 8, ST7735_WHITE); }

} display = tft;

// ------------------------------- Bank setup ------------------------------- //
// ========================================================================== //

/*
   Create a bank and a bank selector to change its setting.
*/

Bank<4> bank(2); // Create a new bank with two tracks per bank

// Create a new bank selector to control the bank using two push buttons
IncrementDecrementSelector<4> bankselector = { bank, {5, 6}, Wrap::Wrap };

// -------------------------- MIDI Input Elements --------------------------- //
// ========================================================================== //

/*
   Define all elements that listen for MIDI messages.
*/

// Time display keeps track of the bar counter
MCU::TimeDisplay timedisplay = {};

// ---------------------------- Display Elements ---------------------------- //
// ========================================================================== //

/*
   Define all display elements that display the state of the input elements.
*/

// Time display
MCU::TimeDisplayDisplay timedisplaydisplay = {
     position (0, 0), font size (1)
    display, timedisplay, {0, 0}, 1, ST7735_WHITE,
};

// --------------------------------- Setup ---------------------------------- //
// ========================================================================== //

void setup() {
    // Correct relative mode for MCU rotary encoders

    tft.initR(INITR_144GREENTAB);

    RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
    Control_Surface.begin(); // Initialize Control Surface
}

// ---------------------------------- Loop ---------------------------------- //
// ========================================================================== //

void loop() {
    Control_Surface.loop(); // Refresh all elements
}

masydoblig, 2020-11-01:
**and this is the Adapter**

#pragma once

#include <Adafruit_ST7735.h> 
#include <Display/DisplayInterface.hpp>

BEGIN_CS_NAMESPACE

/**
 * @brief   This class creates a mapping between the Adafruit_SSD1306 display 
 *          driver and the general display interface used by the Control Surface
 *          library.
 */
class ST7735_DisplayInterface : public DisplayInterface {
  protected:
    ST7735_DisplayInterface(Adafruit_ST7735 &display) : tft(display) {}

public:
    void clear() override { tft.fillScreen(ST77XX_BLACK)(); }
    void drawBackground() override = 0;
    void display() override { tft.initR(); }

    void drawPixel(int16_t x, int16_t y, uint16_t color) override {
        tft.drawPixel(x, y, color);
    }

    void setTextColor(uint16_t color) override { tft.setTextColor(color); }
    void setTextSize(uint8_t size) override { tft.setTextSize(size); }
    void setCursor(int16_t x, int16_t y) override { tft.setCursor(x, y); }

    size_t write(uint8_t c) override { return tft.write(c); }

    void drawLine(int16_t x0, int16_t y0, int16_t x1, int16_t y1,
                  uint16_t color) override {
        tft.drawLine(x0, y0, x1, y1, color);
    }
    void drawFastVLine(int16_t x, int16_t y, int16_t h,
                       uint16_t color) override {
        tft.drawFastVLine(x, y, h, color);
    }
    void drawFastHLine(int16_t x, int16_t y, int16_t w,
                       uint16_t color) override {
        tft.drawFastHLine(x, y, w, color);
    }

    void drawXBitmap(int16_t x, int16_t y, const uint8_t bitmap[], int16_t w,
                     int16_t h, uint16_t color) override {
        tft.drawXBitmap(x, y, bitmap, w, h, color);
    }

  protected:
    Adafruit_ST7735 &tft;
};

END_CS_NAMESPACE

tttapa, 2020-11-01:
I don't have the right hardware to test anything, but this seems strange to me:

```cpp
    void begin() override {
        // Initialize the Adafruit_SSD1306 display
        if (!tft.begin())
            FATAL_ERROR(F("ST7735 allocation failed."), 0x1306);

        // If you override the begin method, remember to call the super class method
        ST7735_DisplayInterface::begin();
    }
```
Looking at the Adafruit examples (https://github.com/adafruit/Adafruit-ST7735-Library/blob/master/examples/graphicstest/graphicstest.ino), I think it should be:

```cpp
    void begin() override {
        tft.initR(INITR_144GREENTAB);
        // If you override the begin method, remember to call the super class method
        ST7735_DisplayInterface::begin();
    }
```
And then don't call `tft.initR(...)` in your setup:

```cpp
void setup() {
    RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
    Control_Surface.begin(); // Initialize Control Surface
}
```

The `DisplayInterface::display()` method writes the display buffer to the display. Since the `Adafruit_ST7735` library writes to the display directly without a buffer, you have to leave the `display()` method empty:

 ```cpp
class ST7735_DisplayInterface : public DisplayInterface {
    ...
    void display() override { /* nothing */ }
    ...
};
```