tttapa/Control-Surface #304: Error al compilar

dagame81, 2020-10-17:
Hello
I have this error When I first Compile an example code that You have, 
This error appears to me in all library examples . 
What solution do you give me? 
Sorry for my English since I an e Spanish and I am translating.
And also sorry for my low teaching, I don't have much idea about this and I'm learning
THANKS for you time
[image]
[image]

tttapa, 2020-10-17:
Please don't post pictures of text, post the actual text. (Especially don't take photos of a computer screen, use the PrintScreen button on your keyboard or a screenshot utility.)

The compilation error is not visible in your photo, only part of it is shown. Please post the full output and the version of Control Surface you're using (release version or git commit hash).

dagame81, 2020-10-18:
Arduino:1.8.2 (Windows 10), Tarjeta:"Arduino/Genuino Uno"

In file included from C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/STL/type_traits:6:0,

                 from C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/Arduino-Wrapper.h:4,

                 from C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/Control_Surface.h:11,

                 from C:\Users\Pablo\AppData\Local\Temp\arduino_modified_sketch_997187\sketch_oct18a.ino:1:

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/STL/Fallback/type_traits:680:66: error: template argument 2 is invalid

     : public integral_constant<bool, __is_trivially_copyable(_Tp)>

                                                                  ^

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/STL/Fallback/type_traits:1365:46: error: template argument 2 is invalid

    __is_trivially_constructible(_Tp, _Args...)>>

                                              ^

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/STL/Fallback/type_traits:1365:47: error: template argument 2 is invalid

    __is_trivially_constructible(_Tp, _Args...)>>

                                               ^

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/STL/Fallback/type_traits:1409:48: error: template argument 2 is invalid

    __is_trivially_constructible(_Tp, const _Tp&)>>

                                                ^

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/STL/Fallback/type_traits:1409:49: error: template argument 2 is invalid

    __is_trivially_constructible(_Tp, const _Tp&)>>

                                                 ^

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/STL/Fallback/type_traits:1417:43: error: template argument 2 is invalid

    __is_trivially_constructible(_Tp, _Tp&&)>>

                                           ^

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/STL/Fallback/type_traits:1417:44: error: template argument 2 is invalid

    __is_trivially_constructible(_Tp, _Tp&&)>>

                                            ^

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/STL/Fallback/type_traits:1425:38: error: template argument 2 is invalid

    __is_trivially_assignable(_Tp, _Up)>>

                                      ^

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/STL/Fallback/type_traits:1425:39: error: template argument 2 is invalid

    __is_trivially_assignable(_Tp, _Up)>>

                                       ^

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/STL/Fallback/type_traits:1433:46: error: template argument 2 is invalid

    __is_trivially_assignable(_Tp&, const _Tp&)>>

                                              ^

[... 67 lines cut ...]

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/Filters/EMA.hpp: In instantiation of 'static constexpr bool EMA<K, input_t, state_t>::supports_range(T, T) [with T = unsigned int; unsigned char K = 2u; input_t = unsigned int; state_t = unsigned int]':

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/Hardware/FilteredAnalog.hpp:217:62:   required from 'class AH::GenericFilteredAnalog<unsigned int (*)(unsigned int), 7u, 2u, unsigned int, unsigned int, 4u>'

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/Hardware/FilteredAnalog.hpp:263:7:   required from 'class AH::FilteredAnalog<7u, 2u, unsigned int, unsigned int, 4u>'

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Outputs/Abstract/MIDIFilteredAnalog.hpp:72:45:   required from 'class CS::MIDIFilteredAnalogAddressable<CS::ContinuousCCSender>'

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/MIDI_Outputs/CCPotentiometer.hpp:20:14:   required from here

C:\Users\Pablo\Documents\Arduino\libraries\Control-Surface-master\src/AH/Filters/EMA.hpp:126:22: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]

                : min >= -sstate_t(max_state >> (K + 1)) - 1 &&  

                      ^

cc1plus.exe: some warnings being treated as errors

exit status 1
Error compilando para la tarjeta Arduino/Genuino Uno.

Este reporte podría tener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

dagame81, 2020-10-18:
That's ok?

dagame81, 2020-10-18:
that error always appears to me

tttapa, 2020-10-18:
That's because the compiler you're using is too old. If the Arduino IDE has a board manager (Tools > board > boards manager) use it to upgrade the Arduino AVR core to version 1.8.3. Otherwise, you'll have to upgrade the IDE.

dagame81, 2020-10-18:
Ok it's solved thank you very much, I will continue investigating to be able to become a dj controller

closed 2020-10-25 by dagame81