tttapa/Control-Surface #305: Arm/Rec switch with led in ableton

kraftZeug, 2020-10-21:
Hello.
I've searched (maybe not enough) in the examples, but didn't found how to code the led input.
I have a bankable switch I use to record loops on ableton and I'd like its led to light up when it's recording.
How could I do that?
This is the code i used for the switch:
`Bankable::CCButton Rec = {
  {bank, BankType::CHANGE_CHANNEL},
  10,
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_6},
  };
`

tttapa, 2020-10-21:
You can use the [`CCValueLED`](https://tttapa.github.io/Control-Surface-doc/Doxygen/da/d34/classCCValueLED.html) class, see the [Getting Started](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d7d/md_pages_Getting-Started.html#first-input) guide.

kraftZeug, 2020-10-21:
Ok, I'm really sorry. I missed the Getting Started guide. :-/
I'll take time to read it now. :-)

closed 2020-10-21 by kraftZeug

kraftZeug, 2020-10-21:
Ok.
I supposed i just had to use the same midi address for the switch and the led, so i wrote this:
`Bankable::CCButton Rec = {
  {bank, BankType::CHANGE_CHANNEL},
  10,
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_6},
  };
  
Bankable::CCValueLED RecLed = {
  {bank, BankType::CHANGE_CHANNEL},
  11,
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_6},
  };
`

And i encountered this error:
`Arduino: 1.8.10 (Linux), TD: 1.48, Board: "Teensy 3.2 / 3.1, Serial + MIDI, 96 MHz (overclock), Faster, US English"

Krafter_V8:126: error: invalid use of template-name 'CS::Bankable::CCValueLED' without an argument list
 Bankable::CCValueLED RecLed = {
 ^
Multiple libraries were found for "SoftwareSerial.h"
 Used: /home/frag/Bureau/arduino-1.8.10/hardware/teensy/avr/libraries/SoftwareSerial
Multiple libraries were found for "Control_Surface.h"
 Used: /home/frag/Arduino/libraries/Control-Surface-master
Multiple libraries were found for "SPI.h"
 Used: /home/frag/Bureau/arduino-1.8.10/hardware/teensy/avr/libraries/SPI
invalid use of template-name 'CS::Bankable::CCValueLED' without an argument list

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

tttapa, 2020-10-21:
You have to specify the bank size as a template parameter:
```cpp
Bank<4> bank;
Bankable::CCValueLED<4> RecLed = {
  {bank, BankType::CHANGE_CHANNEL},
  11,
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_6},
};
```