tttapa/Control-Surface #306: increment selector sons t work with the last control surface

benwadub, 2020-10-23:
hi Pieter u downloaded the last control surface code and now my increment selector won t compile, did you changed something?
**Post your code**

```cpp
IncrementSelector<2> selector = {
    bank,       // Bank to manage
    20,     // push button pins (increment, decrement)
    Wrap::Wrap, // Wrap around
    },
```

tttapa, 2020-10-23:
The `Wrap` argument was removed (because if you don't wrap, there's no way to return to zero).

https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d4c/classIncrementSelector.html#a39a4b853b6810ed11d25c8fbe6e0ec83

benwadub, 2020-10-23:
ok thanks!
i écouter another problem, I try to use the python script to make xbm with a logo to use it at screen startup but when I run the python code in atom I get this answer ```cpp
Traceback (most recent call last):
  File "/Users/benoit/Arduino/libraries/Control-Surface-master/src/Display/Bitmaps/Scripts/XBM-export.py", line 34, in <module>
    os.makedirs(pngoutputdir, exist_ok=True)
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/os.py", line 223, in makedirs
    mkdir(name, mode)
OSError: [Errno 30] Read-only file system: '/xbm'
[Finished in 0.099s]```
do you know what I do wrong?

tttapa, 2020-10-23:
What's your current working directory when executing the script? It looks like you're in the root folder, which you won't have access to as a regular user.

benwadub, 2020-10-23:
Wouah i don t know it s the first time I try to run a python script, I juste found the script in the control surface folder then opened it in atom and ran it using cmd+i, I ll try to find how to see in what folder I am

benwadub, 2020-10-23:
with my new file added in xbm format I got this
```cpp
logo.xbm
Traceback (most recent call last):
  File "/Users/benoit/Arduino/libraries/Control-Surface-master/src/Display/Bitmaps/Scripts/XBM-export.py", line 72, in <module>
    contents = file.read()
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xcc in position 24: invalid continuation byte
[Finished in 0.05s]```
maybe it s a too big xbm?

tttapa, 2020-10-23:
There should be a README with instructions in the `Control-Surface/src/Display/Bitmaps/` folder.

tttapa, 2020-10-23:
If the decoder fails it means that it's not a text file, so probably not a valid XBM file either.

tttapa, 2020-10-23:
See https://github.com/tttapa/Control-Surface/tree/master/src/Display/Bitmaps#readme

benwadub, 2020-10-23:
Yes i alreadu read that, I imported my black and white png in gimp then exported it as a .Xbox then copy paste in the xbm folder that’s after that I get this error

tttapa, 2020-10-23:
Please post the contents of the .xbm file here.

benwadub, 2020-10-23:
I found, it was quite too big!! Thanks for help again!!!

closed 2020-10-23 by benwadub