tttapa/Control-Surface #308: Need help for bank buttons & pots

4dvn, 2020-10-29:
Hi, i need you help for make code of this midi controller
[image]
[image]
But i have the 16 buttons and pots + 1 bank changed button like this, i don't know how to code with banks of button & pot like this explain 
Best regards

tttapa, 2020-10-29:
If I understand your question correctly, this should work:
```cpp
#include <Control_Surface.h>

// Helper for using multiple banks with different offsets with
// a single selector.
template <setting_t N, size_t M>
struct MultiBank : public Selectable<N> {
  MultiBank(const Array<Bank<N>, M> &banks) : banks(banks) {}
  void select(setting_t bankSetting) override {
    for (auto &bank : banks)
      bank.select(bankSetting);
  }
  Bank<N> &operator[](size_t i) { return banks[i]; }
  const Bank<N> &operator[](size_t i) const { return banks[i]; }
  Array<Bank<N>, M> banks;
};

// Two times 4 banks, one has an offset of 4, because the addresses
// of the potentiometers go up by 4 for each bank, and the other has
// an offset of 8, because the addresses of the buttons go up by 8
// for each bank.
MultiBank<4, 2> banks = {{ 4, 8 }};

// The small back button. When pressed, it allows you to change the
// bank by pressing one of the 4 first colored buttons.
Button bankButton = 11;
// The main selector, just selects the bank of the button that's
// pressed.
ManyButtonsSelector<4> selector = {banks, {2, 3, 4, 5}};

Bankable::CCPotentiometer knobs[] = {
  {banks[0], A0, 102},
  {banks[0], A1, 103},
  {banks[0], A2, 104},
  {banks[0], A3, 105},
};

Bankable::NoteButton buttons[] = {
  {banks[1], 2, 52}, // Note that the first 4 NoteButtons use the
  {banks[1], 3, 53}, // same pins as the bank selector
  {banks[1], 4, 54},
  {banks[1], 5, 55},
  {banks[1], 6, 56},
  {banks[1], 7, 57},
  {banks[1], 8, 58},
  {banks[1], 9, 59},
};

USBMIDI_Interface midi;

void setup() {
  bankButton.begin();
  Control_Surface.begin();
  // Disable the bank selector, by default the buttons are used for 
  // the NoteButtons
  Selector<4>::disable(selector);
}

void loop() {
  if (bankButton.update() == Button::Falling) {
    // If the small button is pressed, disable the NoteButtons, and
    // use the buttons to select the bank by enabling the selector.
    Bankable::NoteButton::disable(buttons);
    ManyButtonsSelector<4>::enable(selector);
  } else if (bankButton.getState() == Button::Rising) {
    // If the small button is released, do the reverse, disable the
    // selector and enable the NoteButtons again.
    ManyButtonsSelector<4>::disable(selector);
    Bankable::NoteButton::enable(buttons);
  }
  Control_Surface.loop();
}
```

4dvn, 2020-10-29:
> If I understand your question correctly, this should work:
>
> ```cpp
>
> #include <Control_Surface.h>
>
>
>
> // Helper for using multiple banks with different offsets with
>
> // a single selector.
>
> template [...]
I use the arduino uno with button matrix method to save the pinout, can you update the code for matrix 4x4, thanks you!

tttapa, 2020-10-29:
You cannot use this approach with button matrices, you'd have to roll your own matrix scanning code. I'll see if I can add ExtIO support for button matrices, but I don't have much time.

4dvn, 2020-10-29:
> You cannot use this approach with button matrices, you'd have to roll your own matrix scanning code. I'll see if I can add ExtIO support for button matrices, but I don't have much time.

The arduino mega too big,if you have free time pls update it, i'm making the wood case for matrix ^^, aslo thank you for support me

4dvn, 2020-10-29:
[image]
Oh the pots and buttons like this @tttapa

4dvn, 2020-10-31:
> You cannot use this approach with button matrices, you'd have to roll your own matrix scanning code. I'll see if I can add ExtIO support for button matrices, but I don't have much time.

I really need the matrcies for arduino uno, can you update the library, because the digital and analog pinout are full if not use matrix buttons :(
[image]

tttapa, 2020-11-07:
You could try something like this:  
[ScanningMatrix.zip](https://github.com/tttapa/Control-Surface/files/5504065/ScanningMatrix.zip)

Add it to your `src/AH/Hardware/ExtendedInputOutput` folder and \#include it in your sketch.

4dvn, 2020-11-07:
> You could try something like this:  
> 
> [ScanningMatrix.zip](https://github.com/tttapa/Control-Surface/files/5504065/ScanningMatrix.zip)
> 
> 
> 
> Add it to your `src/AH/Hardware/ExtendedInputOutput` folder and \#include it in your sketch.
> 
> 

Thank you! Can you update the code for 16 buttons matrices bank with this library?

tttapa, 2020-11-08:
You can just use it like the other ExtendedInputOutput elements, create an instance of `ScanningMatrix`, and pass its pins to the `Bankable::NoteButton` constructor.

```cpp
ScanningMatrix<4, 2> matrix {
  {2, 3, 4, 5}, // row pins
  {6, 7},       // column pins
};

Bankable::NoteButton buttons[] = {
  {banks[1], matrix.pin(0), 52}, // Note that the first 4 NoteButtons use the
  {banks[1], matrix.pin(1), 53}, // same pins as the bank selector
  {banks[1], matrix.pin(2), 54},
  {banks[1], matrix.pin(3), 55},
  // ...
};
```
You could add a function to the `ScanningMatrix` to get the pin given a row and column index, instead of a single linear index.

4dvn, 2020-11-08:
> You can just use it like the other ExtendedInputOutput elements, create an instance of `ScanningMatrix`, and pass its pins to the `Bankable::NoteButton` constructor.
> 
> ```c++
> ScanningMatrix<4, 2> matrix {
>   {2, 3, 4, 5}, // row pins
>   {6, 7},       // column pins
> };
> 
> Bankable::NoteButton buttons[] = {
>   {banks[1], matrix.pin(0), 52}, // Note that the first 4 NoteButtons use the
>   {banks[1], matrix.pin(1), 53}, // same pins as the bank selector
>   {banks[1], matrix.pin(2), 54},
>   {banks[1], matrix.pin(3), 55},
>   // ...
> };
> ```
> 
> You could add a function to the `ScanningMatrix` to get the pin given a row and column index, instead of a single linear index.

Oh i got it, also thank you for this feauture

4dvn, 2020-11-09:
> > You can just use it like the other ExtendedInputOutput elements, create an instance of `ScanningMatrix`, and pass its pins to the `Bankable::NoteButton` constructor.
> > ```c++
> > ScanningMatrix<4, 2> matrix {
> >   {2, 3, 4, 5}, // row pins
> >   {6, 7},       // column pins
> > };
> > 
> > Bankable::NoteButton buttons[] = {
> >   {banks[1], matrix.pin(0), 52}, // Note that the first 4 NoteButtons use the
> >   {banks[1], matrix.pin(1), 53}, // same pins as the bank selector
> >   {banks[1], matrix.pin(2), 54},
> >   {banks[1], matrix.pin(3), 55},
> >   // ...
> > };
> > ```
> > 
> > 
> > You could add a function to the `ScanningMatrix` to get the pin given a row and column index, instead of a single linear index.
> 
> Oh i got it, also thank you for this feauture,
> Is that right for 16 buttons? here is my code
> `ScanningMatrix<4, 4> matrix {
> {2, 3, 4, 5}, // row pins
> {6, 7, 8, 9}, // column pins
> };
> 
> Bankable::NoteButton buttons[] = {
> {banks[0], matrix.pin(0), 36}, // Note that the first 4 NoteButtons use the
> {banks[0], matrix.pin(1), 37}, // same pins as the bank selector
> {banks[0], matrix.pin(2), 38},
> {banks[0], matrix.pin(3), 39},
> 
> {banks[0], matrix.pin(4), 40},
> {banks[0], matrix.pin(5), 41}, // Note that the first 4 NoteButtons use the
> {banks[0], matrix.pin(6), 42}, // same pins as the bank selector
> {banks[0], matrix.pin(7), 43},
> 
> {banks[0], matrix.pin(8), 44},
> {banks[0], matrix.pin(9), 45},
> {banks[0], matrix.pin(10), 46}, // Note that the first 4 NoteButtons use the
> {banks[0], matrix.pin(11), 47},
> 
> {banks[0], matrix.pin(12), 48},
> {banks[0], matrix.pin(13), 49},
> {banks[0], matrix.pin(14), 50},
> {banks[0], matrix.pin(15), 51},
> 
> };`

i need help for this :(

tttapa, 2020-11-09:
Please explain the problem and post your full code.

4dvn, 2020-11-09:
> Please explain the problem and post your full code.

Can can't define to map it to 4x4 matrices, ( 16 buttons bank from note 36-51 )
[image]

tttapa, 2020-11-09:
> Can can't define to map it to 4x4 matrices, ( 16 buttons bank from note 36-51 )

That doesn't give me anything to go by.

4dvn, 2020-11-09:
> > Can can't define to map it to 4x4 matrices, ( 16 buttons bank from note 36-51 )
> 
> 
> 
> I can't help you if you don't provide more information.

Oh i mean that how to transfer the old matrix code to make 4 bank matrices code scan:
const AddressMatrix<4, 4> addresses = { {
  {48, 49, 50, 51},
  {44, 45, 46, 47},
  {40, 41, 42, 43},
  {36, 37, 38, 39},
} };

tttapa, 2020-11-09:
If you use `ScanningMatrix`, you have to define each button separately, you cannot use an address matrix. See https://tttapa.github.io/Control-Surface-doc/Doxygen/da/d2d/Transposer_8ino-example.html.

tttapa, 2020-11-09:
But that's almost the same as the code you already had here: https://github.com/tttapa/Control-Surface/issues/308#issuecomment-723583801  
So I don't understand what you're having trouble with?

4dvn, 2020-11-09:
> But that's almost the same as the code you already had here: https://github.com/tttapa/Control-Surface/issues/308#issuecomment-723583801  
> 
> So I don't understand what you're having trouble with?

I have trouble to make it work with 4x4 like this:

ScanningMatrix<4, 4> matrix {
  {2, 3, 4, 5}, // row pins
  {6, 7, 8, 9},       // column pins
};

Bankable::NoteButton buttons[] = {
  {banks[1], matrix.pin(0), 52}, // Note that the first 4 NoteButtons use the
  {banks[1], matrix.pin(1), 53}, // same pins as the bank selector
  {banks[1], matrix.pin(2), 54},
  {banks[1], matrix.pin(3), 55},
  // is that create 16 notes like this? " which is that "matrix.pin(2)" mean on digital pin of arduino or pin of rows & collums?
};

tttapa, 2020-11-09:
Yes, you have to declare all 16 notes like that.  
`matrix.pin(2)` is the switch in the third column of the first row, you can see the calculation in the `ScanningMatrix.hpp` file I posted.

4dvn, 2020-11-09:
> i need help for this :(

So is that right? ^^ i added to 16 notes but not sure the pin()

tttapa, 2020-11-09:
It looks alright at first sight. Why don't you try it? Does it compile? Does it send the note events you expect.

4dvn, 2020-11-09:
> It looks alright at first sight. Why don't you try it? Does it compile? Does it send the note events you expect?

Is show that maybe out of memory on arduino

tttapa, 2020-11-09:
You could try disabling the SysEx input, which should save you over 120 bytes of RAM. You do need the latest patch (f351bffd3a674cd68998a8900daa1a45fc257615). Set the following setting to `1`:
https://github.com/tttapa/Control-Surface/blob/f351bffd3a674cd68998a8900daa1a45fc257615/src/Settings/Settings.hpp#L47-L48

You could also try marking the `UpdatableCRTP::applyToAll` methods `always_inline`, like this:
https://github.com/tttapa/Control-Surface/blob/71cc73370d32c5a2a7ad94bb162299ef3fadf5be/src/AH/Containers/Updatable.hpp#L84-L96

If that's not enough, you'll either have to accept that the UNO is not a suitable board for your project, or roll your own optimized code. Abstractions like the `ScanningMatrix` class I sent you are easy to use, but this comes at a cost.

brosekerg, 2021-02-11:
this code works for 12 buttons. its going to be 16. Im sorry for jumping on this post but im unaware on how to get publicity for mine. i also need help in making a 4x4 button matrix. i want 16 buttons for notes and a stepped pot that has 4 banks. could you show me how to do this? its most likely a sendProgramChange i just need to know how to build the arrays that hold the banks and their patches in each bank. could you help me?

#include <MIDI.h>

MIDI_CREATE_DEFAULT_INSTANCE();

int middleCpin = 2;  
int cSharpPin = 3;   
int middleDpin = 4;  
int dSharpPin= 5;
int middleEpin = 6;
int middleFpin = 7;
int fSharpPin = 8;
int middleGpin = 9;
int gSharpPin = 10;
int middleApin = 11;
int aSharpPin = 12;
int middleBpin = 13;

int analogpot1 = A0;  //knob 1
int analogpot2 = A1;  //knob 2

int analogpot1Old = 0;
int analogpot2Old = 0;
int analogpot1New = 0;
int analogpot2New = 0;

#define analogpot1CC 54
#define analogpot2CC 55

void setup() {
  // put your setup code here, to run once:
MIDI.begin (); // MIDI START

pinMode(middleCpin, INPUT_PULLUP);
pinMode(cSharpPin, INPUT_PULLUP);
pinMode(middleDpin, INPUT_PULLUP);
pinMode(dSharpPin, INPUT_PULLUP);
pinMode(middleEpin, INPUT_PULLUP);
pinMode(middleFpin, INPUT_PULLUP);
pinMode(fSharpPin, INPUT_PULLUP);
pinMode(middleGpin, INPUT_PULLUP);
pinMode(gSharpPin, INPUT_PULLUP);
pinMode(middleApin, INPUT_PULLUP);
pinMode(aSharpPin, INPUT_PULLUP);
pinMode(middleBpin, INPUT_PULLUP);

pinMode(analogpot1, INPUT);
pinMode(analogpot2, INPUT);

//Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:

static bool middleCvalueOld = HIGH;
static bool cSharpValueOld = HIGH;
static bool middleDvalueOld = HIGH;
static bool dSharpValueOld = HIGH;
static bool middleEvalueOld = HIGH;
static bool middleFvalueOld = HIGH;
static bool fSharpValueOld = HIGH;
static bool middleGvalueOld = HIGH;
static bool gSharpValueOld = HIGH;
static bool middleAvalueOld = HIGH;
static bool aSharpValueOld = HIGH;
static bool middleBvalueOld = HIGH;

//footswitches
[... 163 lines cut ...]

    if (analogpot1New - analogpot1Old >= 35 || analogpot1Old - analogpot1New >= 35) {
      analogpot1Old = analogpot1New;
      analogpot1New = (map(analogpot1New, 1023, 0, 0, 120));
      analogpot1New = (constrain(analogpot1New, 0, 120));
      MIDI.sendControlChange(analogpot1CC, analogpot1New, 1);
//      Serial.print ("pot: ");
//      Serial.println(pot1);
//      Serial.print("potread: ");
//      Serial.println(analogpot1New); 
    }
    
    if (analogpot2New - analogpot2Old >= 35 || analogpot2Old - analogpot2New >= 35) {
      analogpot2Old = analogpot2New;
      analogpot2New = (map(analogpot2New, 1023, 0, 0, 120));
      analogpot2New = (constrain(analogpot2New, 0, 120));
      MIDI.sendControlChange(analogpot2CC, analogpot2New, 1);
//      Serial.print ("pot: ");
//      Serial.println(pot2);
//      Serial.print("potread: ");
//      Serial.println(analogpot2New); 
    }

delay(25);
}

DRCRecoveryData, 2024-04-20:
> this code works for 12 buttons. its going to be 16. Im sorry for jumping on this post but im unaware on how to get publicity for mine. i also need help in making a 4x4 button matrix. i want 16 [...]

```
#include <MIDI.h>
MIDI_CREATE_DEFAULT_INSTANCE();

// Pins for the 16 buttons
int buttonPins[] = {2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17};

// Pins for the stepped potentiometer
int potPin = A0;

// Define MIDI program changes for each bank
int bankPrograms[4][16] = {
    // Bank 1
    {0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15},
    // Bank 2
    {16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31},
    // Bank 3
    {32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47},
    // Bank 4
    {48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63}
};

int currentBank = 0;

void setup() {
    MIDI.begin();

    // Set button pins as input with internal pull-up resistors
    for (int i = 0; i < 16; i++) {
        pinMode(buttonPins[i], INPUT_PULLUP);
    }
}

void loop() {
    // Read potentiometer value and determine the current bank
    int potValue = analogRead(potPin);
    currentBank = map(potValue, 0, 1023, 0, 3);

    // Scan button matrix and handle MIDI program changes
    for (int i = 0; i < 16; i++) {
        if (digitalRead(buttonPins[i]) == LOW) {
            int patchNumber = bankPrograms[currentBank][i];
            if (patchNumber != -1) {  // Check if a valid patch is assigned
                MIDI.sendProgramChange(patchNumber, 1);
            }
            delay(100);  // Add debounce delay if necessary
        }
    }
}
```