tttapa/Control-Surface #309: Platformio error "MIDIUSB can only be used with an USB MCU"

xabadabazinho, 2020-10-29:
Hi, thanks for the fastastic library. I´m compiling without problems in Arduino IDE, but having erros in Platformio/VSC using BLE with ESP32 and Control Surface@^1.2.0-4:

Platformio.ini

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps = 
	tttapa/Control Surface@^1.2.0-4
monitor_speed = 115200

=====================================================
The code 

#include <Arduino.h>
#include <Control_Surface.h>
#include <MIDI_Interfaces/BluetoothMIDI_Interface.hpp>

BluetoothMIDI_Interface midi;

void setup()
{
    Serial.begin(115200);
    midi.begin();
}

using namespace MIDI_Notes;

const MIDIAddress address = {note(C, 4), CHANNEL_1};
const uint8_t velocity = 0x7F;

void loop()
{
    midi.sendNoteOn(address, velocity);
    midi.sendNoteOff(address, velocity);
    delay(500);
    midi.update(); // Handle or discard MIDI input
}

=======================================================
errors:

In file included from .pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.cpp:17:0:
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.h:18:2: error: #error MIDIUSB can only be used with an USB MCU.
 #error MIDIUSB can only be used with an USB MCU.
  ^
In file included from .pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.cpp:17:0:
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.h:74:2: error: #error "Unsupported architecture"
 #error "Unsupported architecture"
Compiling .pio\build\esp32doit-devkit-v1\libc83\BLE\BLE2902.cpp.o
  ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.h:125:2: error: 'EndpointDescriptor' does not name a type
  EndpointDescriptor len;  // 9
  ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.h:154:2: error: 'IADDescriptor' does not name a type     
  IADDescriptor                      iad;
  ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.h:156:2: error: 'InterfaceDescriptor' does not name a type
  InterfaceDescriptor                Audio_ControlInterface;
  ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.h:160:2: error: 'InterfaceDescriptor' does not name a type
  InterfaceDescriptor                Audio_StreamInterface;
  ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.h:208:1: error: expected class-name before '{' token
 {
 ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.h:213:2: error: 'EPTYPE_DESCRIPTOR_SIZE' does not name a 
type
  EPTYPE_DESCRIPTOR_SIZE epType[2];   ///< Container that defines the two bulk MIDI IN/OUT endpoints types
[... 168 lines cut ...]
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.cpp:164:30: error: 'is_write_enabled' was not declared in this scope
  if (is_write_enabled(MIDI_TX))
                              ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.cpp:167:41: error: 'USB_Send' was not declared in this scope
   int r = USB_Send(MIDI_TX, buffer, size);
                                         ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.cpp: In constructor 'MIDI_::MIDI_()':
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.cpp:190:22: error: class 'MIDI_' does not have any field 
named 'PluggableUSBModule'
 MIDI_::MIDI_(void) : PluggableUSBModule(2, 2, epType)
                      ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.cpp:190:47: error: 'epType' was not declared in this scope
 MIDI_::MIDI_(void) : PluggableUSBModule(2, 2, epType)
                                               ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.cpp:192:14: error: 'EP_TYPE_BULK_OUT_MIDI' was not declared in this scope
  epType[0] = EP_TYPE_BULK_OUT_MIDI; // MIDI_ENDPOINT_OUT
              ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.cpp:193:14: error: 'EP_TYPE_BULK_IN_MIDI' was not declared in this scope
  epType[1] = EP_TYPE_BULK_IN_MIDI;  // MIDI_ENDPOINT_IN
              ^
.pio\libdeps\esp32doit-devkit-v1\MIDIUSB\src\MIDIUSB.cpp:194:15: error: 'PluggableUSB' was not declared in this scope
  PluggableUSB().plug(this);
               ^
*** [.pio\build\esp32doit-devkit-v1\lib813\MIDIUSB\MIDIUSB.cpp.o] Error 1
======================================== [FAILED] Took 44.53 seconds ========================================

tttapa, 2020-10-29:
I'm not a PlatformIO user myself, but it looks like it's trying to build the MIDIUSB library, which shouldn't be necessary for ESP32. I'm not sure what causes this, it might be the dependencies specified in Control Surface's library.json file, or it might detect it on its own by scanning the header files used in the code using the pio LDF.

tttapa, 2020-10-29:
Adding `lib_ignore = MIDIUSB` to your `platformio.ino` file should get around this issue. 

I'm not sure how to fix this permanently, though, by changing the dependencies in Control Surface's `library.json` file, for example, so if you have more experience with this, feel free to let me know!

xabadabazinho, 2020-10-29:
Works perfectly Thanks a lot.

tttapa, 2020-10-29:
On an unrelated note: there have been huge improvements in the ESP32 Bluetooth MIDI backend, I've added SysEx support and replaced the Arduino BLE library with the native ESP-IDF Bluetooth API because the former had concurrency issues. The changes haven't made it into the master branch yet, but you can check them out on the [`new-input`](https://github.com/tttapa/Control-Surface/tree/new-input) branch if you want.

emalihin, 2021-03-15:
Suggestion to add `lib_ignore = MIDIUSB` worked well for Teensy 4.0 as well

JeffyOLOLO, 2023-01-18:
Hi, it's still reproduced