tttapa/Control-Surface #310: power distribution and erratic signals

skyriderone, 2020-10-29:
Thanks as always for your amazing work.

I have built my most ambitious controller yet, and everything functions perfectly... until I energize the panel mounted LEDS (inside buttons), and the inlaid LED strips which backlight the controller from inside.  The surface of the controller is Acrylic.

[image]

I'm hoping for troubleshooting tips.

The controller is 3 separate circuits.  The Teensy 3.6 is powered by USB, and I am using GND - not analog ground.  Connected to it are a 4x5 button matrix with 4 banks, 16 pots via a cd4067 mux, 4 note buttons, and 2 IR distance sensors (CCpotentiometer 1,2 in code).  Everything works as expected when I do not power up the lights. 

The panel mounted buttons all have leds inside, which are powered by a 12v 30A DC external power supply.  They all work.

The inlaid LED strips are another circuit, which includes an Arduino Leonardo and a Sparkfun spectrum shield.  The LED strips vary their brightness and color in response to amplitude and frequency of an input audio signal (the audio output from ableton live).  This circuit is powered by another output on the external 12v 30A power supply.  This system works as expected.

It all works together, however, when the external power supply is powered up, energizing the LEDS, I get erratic readings - unintended midi CC messages varying from 00 - 01 in MIDIOX.  It appears to be affecting four potentiometers in particular (0A, 0B, 0C, 0D)   

## Steps to reproduce the problem
The unwanted midi messages are only sent went the external power supply is turned on.

### Hardware
teensy 3.6
  
**Schematic**: ?
I haven't created a schematic, but can if it will help.  

### Software versions:

**MIDI Controller library**: 3.1.1
Control-Surface-master  
**Arduino IDE**:  1.8.12  
**Operating System**: Windows
**Operating System version**: 7
(**Teensyduino**): 1.51
(**Encoder library**): not sure if i have this library  
(**MIDIUSB library**): ? not sure what version... 

### Settings in the IDE
usb type midi

### Full code 
cobbled together from your examples, works as expected.

```cpp
// 

#include <Control_Surface.h>

USBMIDI_Interface midi;

using namespace MIDI_Notes;

// Instantiate a NoteButton object
NoteButton button1 = {
  9,                       // Push button on pin 9
  {note(C, 5), CHANNEL_2}, // Note C5 on MIDI channel 2
};

NoteButton button2 = {
  10,                       // Push button on pin 10
  {note(D, 5), CHANNEL_2}, // Note D5 on MIDI channel 2
};

NoteButton button3 = {
  11,                       // Push button on pin 11
  {note(E, 5), CHANNEL_2}, // Note E5 on MIDI channel 2
};

NoteButton button4 = {
  12,                       // Push button on pin 12
  {note(F, 5), CHANNEL_2}, // Note F5 on MIDI channel 2
};

Bank<4> bank = {20}; // 4 banks, 20 addresses per bank

IncrementDecrementSelectorLEDs<4> bankSelector = {
  bank,
  {38, 39},         // button pins
  {24, 25, 26, 27}, // LED pins
};

AddressMatrix<5, 4> notes = {{
  {1, 2, 3, 4},
  {5, 6, 7, 8},
  {9, 10, 11, 12},
  {13, 14, 15, 16},
  {17, 18, 19, 20},
  }};

Bankable::NoteButtonMatrix<5, 4> buttons = {
  bank,      //
  {0, 1, 2, 3, 4},    // row pins (outputs, driven low-Z low !)
  {5, 6, 7, 8}, // column pins (inputs, hi-Z)
  notes,     //
  CHANNEL_1, //
};

CD74HC4067 mux = { 
  A4,               // analog pin
  {20, 21, 22, 23},  // Address pins S0, S1, S2, S3
  // 13, // Optionally, specify the enable pin
};

// Create an array of potentiometers that send out
// MIDI Control Change messages when you turn the
// potentiometers connected to the eight input pins of
// the multiplexer
CCPotentiometer potentiometers[] = {
  {mux.pin(0), {10, CHANNEL_1}},
  {mux.pin(1), {11, CHANNEL_1}},
  {mux.pin(2), {12, CHANNEL_1}},
  {mux.pin(3), {13, CHANNEL_1}},
  {mux.pin(4), {14, CHANNEL_1}},
  {mux.pin(5), {15, CHANNEL_1}},
  {mux.pin(6), {16, CHANNEL_1}},
  {mux.pin(7), {17, CHANNEL_1}},
  {mux.pin(8), {18, CHANNEL_1}},
  {mux.pin(9), {19, CHANNEL_1}},
  {mux.pin(10), {20, CHANNEL_1}},
  {mux.pin(11), {21, CHANNEL_1}},
  {mux.pin(12), {22, CHANNEL_1}},
  {mux.pin(13), {23, CHANNEL_1}},
  {mux.pin(14), {24, CHANNEL_1}},
  {mux.pin(15), {25, CHANNEL_1}},
};

// Create a new instance of the class `CCPotentiometer`, called `potentiometer`,
// on pin A0, that sends MIDI messages with controller 7 (channel volume)
// on channel 1.
CCPotentiometer potentiometer1 = {
  A0, {MIDI_CC::Channel_Volume, CHANNEL_1}
};

CCPotentiometer potentiometer2 = {
  A1, {MIDI_CC::Channel_Volume, CHANNEL_2}
};
// The filtered value read when potentiometer is at the 0% position
constexpr analog_t minimumValue = 5000;
// The filtered value read when potentiometer is at the 100% position
constexpr analog_t maximumValue = 11000;

// A mapping function to eliminate the dead zones of the potentiometer:
// Some potentiometers don't output a perfect zero signal when you move them to
// the zero position, they will still output a value of 1 or 2, and the same
// goes for the maximum position.
analog_t mappingFunction(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

// :::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: //

void setup() {
    // Add the mapping function to the potentiometer
    potentiometer1.map(mappingFunction);
    potentiometer2.map(mappingFunction);
    // Initialize everything
   Control_Surface.begin();
  
}

void loop() {
  Control_Surface.loop();

}
```

## Steps taken to try to diagnose or solve the problem
I'm not sure how to use midi debug mode, but am using MIDIOX to diagnose.

I connected the ground of all three circuits (teensy, panel leds, led strips).  Is this recommended or ill advised?

I shortened all cables as much as possible.

I replaced the mux and made sure new solder connections were secure and clean. 

I did not use twisted pairs and, while I tried to separate cables, there are cable paths that run parallel in a cluster at times.

I've read similar issues here with the mux, but I'm wondering if there are general practice basics that I may be missing regarding power distribution.

## The goal of your project and aditional information
This is a performance tool for a theatrical production.  I want it to be functional and look cool as well.

Thanks again!

tttapa, 2020-10-29:
I'm afraid I can't really help you with the hardware side of things, you'll be much better off asking that on the Arduino or PJRC forums, maybe even All About Circuits or similar.

What you could try is increasing the amount of digital filtering performed by Control Surface, see https://github.com/tttapa/Control-Surface/issues/297#issuecomment-706700657. That being said, fundamentally you have a bunch of long wires acting as antennas, close and parallel to high currents that are switched at high frequencies, so there's only so much you can do in software. 

The controller looks really cool, I wish you the best of luck with your project!  
Feel free to drop a link here if you decide to open a thread on an online forum.

underwoodblog, 2020-10-29:
What about using shielded cables for the Potentiometers that cause the error?

skyriderone, 2020-10-30:
Thanks for your replies!

Good idea about rewiring the just the trouble pots.  Before adjusting the (flawed) fundamentals of my design, I thought I'd see what could be done with digital filtering.

I followed TTTAPA's advice and changed the analog shift factor to "3" and the filter type to "32".

So far that adjustment has at least cleaned up my output and everything works as it should.

My IR sensors are behaving erratic now, however, and this has never been a problem before.  Do I need to recalibrate them after adjusting the setting.hpp file?  Would that have effected the filter mapping I applied to those CCpotentiometers in the above code?

Thanks!

tttapa, 2020-10-30:
> My IR sensors are behaving erratic now, however, and this has never been a problem before. Do I need to recalibrate them after adjusting the setting.hpp file? Would that have effected the filter mapping I applied to those CCpotentiometers in the above code?

Yes, you have to update the mapping functions, their scaling changes when you change the filter settings. You can use `FilteredAnalog<>::getMaxRawValue()` to get the maximum value of the raw input to the mapping function. For the default setting (filter shift factor = 2 and filter type = `uint16_t`), this will return 16383, for example. You'll have to update your `minimumValue`, `maximumValue` and the constant `16383` in your `mappingFunction`.

I should probably do it correctly in the examples, without the need for any "magic numbers".

skyriderone, 2020-10-31:
> You can use FilteredAnalog<>::getMaxRawValue() to get the maximum value of the raw input to the mapping function.

I was hoping I'd understand this after coffee, but I'm unsure how to use this function.

I understand how to use the serial monitor to find my min and max values, but can you elaborate on how to find the new max value of the raw input please?

tttapa, 2020-10-31:
```cpp
// The filtered value read when potentiometer is at the 0% position
constexpr analog_t minimumValue = 5000ul * FilteredAnalog<>::getMaxRawValue() / 16384;
// The filtered value read when potentiometer is at the 100% position
constexpr analog_t maximumValue = 11000ul * FilteredAnalog<>::getMaxRawValue() / 16384;

analog_t mappingFunction(analog_t raw) {
    Serial.print("Filtered: ");
    Serial.println(raw);
    raw = constrain(raw, minimumValue, maximumValue);
    return map(raw, minimumValue, maximumValue, 0, FilteredAnalog<>::getMaxRawValue());
}
```
I used your existing values and scaled them. Note the `ul` suffix (unsigned long) to prevent overflow.

skyriderone, 2020-11-01:
Awesome.  Thanks.  I replaced the mapping function in the previous code with your new scaling and everything seems to work reliably now.  If I find that I need to further increase the filter shift factor in the future, this code should eliminate the need for updating the IR sensors with new values, correct?

Also, if I may ask... I'm planning on using this controller and another one that I've already finished (4 button xbee wireless) on the same computer (both USB powered).  They work together in Ableton perfectly when I test them on separate usb ports, but I'm planning on using a powered USB hub for the final project.  Are there any known issues using USB hubs?

tttapa, 2020-11-01:
> If I find that I need to further increase the filter shift factor in the future, this code should eliminate the need for updating the IR sensors with new values, correct?

Correct.

> Also, if I may ask... I'm planning on using this controller and another one that I've already finished (4 button xbee wireless) on the same computer (both USB powered). They work together in Ableton [...]

Not that I know of, I use a really cheap 4-port USB hub and I've never encountered any problems. If you do have an issue with your specific setup, it's best to check the PJRC forums, the USB backend used by Control Surface is included in the Teensy Core libraries, it's not part of Control Surface itself.

skyriderone, 2020-11-01:
I can't thank you enough for your work on this library and patience for questions.  It's been a fun learning experience, and has honestly helped keep me sane through unemployment.

Now that I've got a functioning device, I'll bring my hardware questions to PJRC forums.

closed 2020-11-01 by skyriderone