tttapa/Control-Surface #314: Supported chips?

digimbyte, 2020-11-04:
Hi, Looking at making my own Midi Controller when I came across yours while looking up an issue with another DIY project 
(nerd musician & notes and volts DIY school scams)
But I was wondering if anyone has got this working on something on the following:
MEGA 32U4 pro micro
MEGA 328P Pro mini
MEGA 4809 nano-every

I ran into memory stability options with other chips/projects

digimbyte, 2020-11-04:
I also have quite a few ESP12E's that I ended up with when trying to bargain hunt for an ESP8266
would be great if I could use them as well.

tttapa, 2020-11-04:
The ATmega32U4, ATmega328P and the ATmega4809 are all supported by this library. Since they are all 8-bit AVR chips with a very limited amount of RAM, you might run into memory issues if you try to build a very complicated controller.  
The main objective of Control Surface is ease of use, so you'll always be able to come up with a hand-coded solution with a smaller memory footprint. However, for basic controllers with one or two dozen of potentiometers and buttons without any displays, an AVR chip will work just fine.

For anything else, I'd highly recommend using a more powerful ARM (Teensy or SAMD, for example) or ESP32 chip.
The ESP8266 is also supported, but the disadvantage is that it doesn't have a very useful ADC.

digimbyte, 2020-11-04:
ADC?

I found ESP32 examples but so far haven't been able to compile anything for the ESP8266's
my goal is to handle 3 mux's with analog and buttons (checking high/low) and calling the corresponding midi call

digimbyte, 2020-11-04:
pastebin of error [https://pastebin.com/kF49qmHN](https://pastebin.com/kF49qmHN)

tttapa, 2020-11-04:
>  just almost no documentation on it

Most examples work just fine on the ESP8266, there isn't anything special about it, only the ADC is cumbersome, but you find plenty of documentation for the ESP8266 online.

> pastebin of error https://pastebin.com/kF49qmHN

I haven't been able to reproduce these errors, it compiles just fine for me using the ESP8266 core version 2.7.4 and Control Surface master or 1.2.0.  
It looks like it's not compiling any of the library implementation files. Did you install it correctly? Please see the [installation instructions](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/da8/md_pages_Installation.html).

digimbyte, 2020-11-04:
ah, you are right, I accidently imported the src folder, not the whole repo from github

closed 2020-11-04 by digimbyte