tttapa/Control-Surface #315: Extraer direccion de una matrix

alomdj00, 2020-11-05:
Hola Peter.
He cobstruido una matrix de notas de 4x4, con 4 bancos, funciona perfecto con el ejemplo que trae la libreria, sin embargo me gustaria poder visualizar en un oled la nota que he presionado,  he intentado obtener las direcciones de la matrix pero no me ha sido posible, sibpudieras orientarme un poco te lo agradeceria.
Un saludo y abrazo desde España.

tttapa, 2020-11-05:
You can check which buttons are pressed using the [NoteButtonMatrix::getPrevState()](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/daf/classNoteButtonMatrix.html#a62dac074b3767f9be91bc05f4e5a0638) method. The [addresses are private](https://github.com/tttapa/Control-Surface/blob/0efe4a882766894e4e82d5736e4927e92b9d7f41/src/MIDI_Outputs/Abstract/MIDIButtonMatrix.hpp#L61), but you could add a getter for them.
```cpp
template <class Sender, uint8_t nb_rows, uint8_t nb_cols>
class MIDIButtonMatrix : public MIDIOutputElement,
                         public AH::ButtonMatrix<nb_rows, nb_cols> {
    ...
  public:
    uint8_t getAddress(uint8_t row, uint8_t col) const { return adresses[row][col]; }

  private:
    AddressMatrix<nb_rows, nb_cols> addresses;

    ...
};
```

alomdj00, 2020-11-05:
Hola. He estado tratando y tratando y no me ha salido, soy aficionado por lo que no se mucho. Lo mas cercano que me salio fue esto. Solo que en el monitor serie me salen caracteres que no se de donde salen, bien este es lo que hice

``#include <Control_Surface.h>

USBMIDI_Interface midi;

BEGIN_CS_NAMESPACE

namespace Bankable {

template <class BankAddress, class Sender, uint8_t nb_rows, uint8_t nb_cols>
class MIDIButtonMatri : public MIDIOutputElement,
                         public AH::ButtonMatrix<nb_rows, nb_cols> {

  protected:
    MIDIButtonMatri(BankAddress bankAddress,
                     const PinList<nb_rows> &rowPins,
                     const PinList<nb_cols> &colPins, const Sender &sender)
        : AH::ButtonMatrix<nb_rows, nb_cols>(rowPins, colPins),
          address{bankAddress}, sender{sender} {}

  public:
    void begin() override { Serial.begin(115200); AH::ButtonMatrix<nb_rows, nb_cols>::begin(); }

    void update() override { AH::ButtonMatrix<nb_rows, nb_cols>::update(); }

    uint8_t getAddress(uint8_t row, uint8_t col) const {return addresses[row][col];}
    
  private:
    void onButtonChanged(uint8_t row, uint8_t col, bool state) final override {
        if (state == LOW) {
            if (!activeButtons)
                address.lock(); // Don't allow changing of the bank setting
            activeButtons++;
            sender.sendOn(address.getActiveAddress(row, col));
             Serial.println(address.getBaseAddress(row, col));
        } else {
            sender.sendOff(address.getActiveAddress(row, col));
            activeButtons--;
            if (!activeButtons)
                address.unlock();
        }
    }
    Sender sender;
    BankAddress address;
    uint16_t activeButtons = 0;
    AddressMatrix<nb_rows, nb_cols> addresses;
 // public:
    
};

} // namespace Bankable

END_CS_NAMESPACE

BEGIN_CS_NAMESPACE

namespace Bankable {

template <uint8_t nb_rows, uint8_t nb_cols>
class NoteButtonMatri
    : public MIDIButtonMatri<MatrixAddress<nb_rows, nb_cols>,
                              DigitalNoteSender, nb_rows, nb_cols> {
  public:

    NoteButtonMatri(OutputBankConfig<> config, const PinList<nb_rows> &rowPins,
                     const PinList<nb_cols> &colPins,
                     const AddressMatrix<nb_rows, nb_cols> &notes,
                     MIDIChannelCable channelCN = {CHANNEL_1, CABLE_1},
                     uint8_t velocity = 0x7F)
        : MIDIButtonMatri<MatrixAddress<nb_rows, nb_cols>, DigitalNoteSender,
                           nb_rows, nb_cols>{
              {config, notes, channelCN},
              rowPins,
              colPins,
              {velocity},
          } {}

    /// Set the velocity of the MIDI Note events.
    void setVelocity(uint8_t velocity) { this->sender.setVelocity(velocity); }
    /// Get the velocity of the MIDI Note events.
    uint8_t getVelocity() const { return this->sender.getVelocity(); }
};

} // namespace Bankable

END_CS_NAMESPACE

Bank<4> bank = {6}; // 4 banks, 6 addresse per banks
IncrementDecrementSelector<4> selector = {bank, {12, 13}, Wrap::Wrap};

AddressMatrix<1, 2> notes = {{
  {50,60},
}};

Bankable::NoteButtonMatri<1, 2> buttons = {
  bank,      //
  {23},    // row pins (outputs, driven low-Z low !)
  {22,15}, // column pins (inputs, hi-Z)
  notes,     //
  CHANNEL_1, //
};

void setup() {
  Serial.begin(115200);
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
 
}
``

tttapa, 2020-11-06:
It's not clear to me what you're trying to do. Please explain what you expect the code to do, and what it actually does. If you get output on the serial monitor that you don't understand, post it here as well.

alomdj00, 2020-11-07:
> Hola. He estado tratando y tratando y no me ha salido, soy aficionado por lo que no se mucho. Lo mas cercano que me salio fue esto. Solo que en el monitor serie me salen caracteres que no se de donde salen, bien este es lo que hice
> 
> ```c++
> #include <Control_Surface.h>
> 
> USBMIDI_Interface midi;
> 
> 
> BEGIN_CS_NAMESPACE
> 
> namespace Bankable {
> 
> template <class BankAddress, class Sender, uint8_t nb_rows, uint8_t nb_cols>
> class MIDIButtonMatri : public MIDIOutputElement,
>                          public AH::ButtonMatrix<nb_rows, nb_cols> {
> 
>   protected:
>     MIDIButtonMatri(BankAddress bankAddress,
>                      const PinList<nb_rows> &rowPins,
>                      const PinList<nb_cols> &colPins, const Sender &sender)
>         : AH::ButtonMatrix<nb_rows, nb_cols>(rowPins, colPins),
>           address{bankAddress}, sender{sender} {}
> 
>   public:
>     void begin() override { Serial.begin(115200); AH::ButtonMatrix<nb_rows, nb_cols>::begin(); }
> 
>     void update() override { AH::ButtonMatrix<nb_rows, nb_cols>::update(); }
> 
>     uint8_t getAddress(uint8_t row, uint8_t col) const {return addresses[row][col];}
>     
>   private:
>     void onButtonChanged(uint8_t row, uint8_t col, bool state) final override {
>         if (state == LOW) {
>             if (!activeButtons)
>                 address.lock(); // Don't allow changing of the bank setting
>             activeButtons++;
>             sender.sendOn(address.getActiveAddress(row, col));
>              Serial.println(address.getBaseAddress(row, col));
>         } else {
>             sender.sendOff(address.getActiveAddress(row, col));
>             activeButtons--;
>             if (!activeButtons)
>                 address.unlock();
>         }
>     }
>     Sender sender;
>     BankAddress address;
>     uint16_t activeButtons = 0;
>     AddressMatrix<nb_rows, nb_cols> addresses;
>  // public:
>     
> };
> 
> } // namespace Bankable
> 
> END_CS_NAMESPACE
> 
> 
> 
> 
> BEGIN_CS_NAMESPACE
> 
> namespace Bankable {
> 
> template <uint8_t nb_rows, uint8_t nb_cols>
> class NoteButtonMatri
>     : public MIDIButtonMatri<MatrixAddress<nb_rows, nb_cols>,
>                               DigitalNoteSender, nb_rows, nb_cols> {
>   public:
> 
>     NoteButtonMatri(OutputBankConfig<> config, const PinList<nb_rows> &rowPins,
>                      const PinList<nb_cols> &colPins,
>                      const AddressMatrix<nb_rows, nb_cols> &notes,
>                      MIDIChannelCable channelCN = {CHANNEL_1, CABLE_1},
>                      uint8_t velocity = 0x7F)
>         : MIDIButtonMatri<MatrixAddress<nb_rows, nb_cols>, DigitalNoteSender,
>                            nb_rows, nb_cols>{
>               {config, notes, channelCN},
>               rowPins,
>               colPins,
>               {velocity},
>           } {}
> 
>     /// Set the velocity of the MIDI Note events.
>     void setVelocity(uint8_t velocity) { this->sender.setVelocity(velocity); }
>     /// Get the velocity of the MIDI Note events.
>     uint8_t getVelocity() const { return this->sender.getVelocity(); }
> };
> 
> } // namespace Bankable
> 
> END_CS_NAMESPACE
> 
> 
> 
> 
> 
> 
> Bank<4> bank = {6}; // 4 banks, 6 addresse per banks
> IncrementDecrementSelector<4> selector = {bank, {12, 13}, Wrap::Wrap};
> 
> AddressMatrix<1, 2> notes = {{
>   {50,60},
> }};
> 
> Bankable::NoteButtonMatri<1, 2> buttons = {
>   bank,      //
>   {23},    // row pins (outputs, driven low-Z low !)
>   {22,15}, // column pins (inputs, hi-Z)
>   notes,     //
>   CHANNEL_1, //
> };
> 
> 
> 
> void setup() {
>   Serial.begin(115200);
>   Control_Surface.begin();
> }
> 
> void loop() {
>   Control_Surface.loop();
>  
> }
> ```

Mi idea es que cuando presione una tecla de la matriz me muestre en una pantalla el numero de nota, el problema que he encontrado es el no poder obtener el numero de tecla presionado, con el codigo que he publicado funciona a medias, puesto que en el monitor serie me salen valores como estos >? >? 60, siempre me sale esto >? >?, el numero me sale al final pero no puedo meterlo en una variable para mostrarlo en cualquier pantalla. Me gustaria que muestre solo el numero, es decir la nota en nunero digamos 60,61,62,63,64.....  No logre incluir o llamar al getPrevState(), me tira error de que no pertenece al notebuttonMatrix

tttapa, 2020-11-08:
The output you're seeing in the Serial monitor is the MIDI output. The `USBMIDI_Interface` class sends serial data if your board doesn't support MIDI over USB, but the MIDI data is not text, so you can't read it in the Serial monitor. You could try using `USBDebugMIDI_Interface` to see the MIDI messages in the serial monitor.