tttapa/Control-Surface #318: Runtime Loop transpose

digimbyte, 2020-11-06:
I noticed a pattern with the library where most things are declared and then nothing is done in the loop
I need to transpose based on the input of a potentiometer, but I can't find out how to shift the transpose without using a pin map
example:
```
Transposer<-12, +12> transposer;
IncrementDecrementSelector<transposer.getNumberOfBanks()> selector = {
  transposer,
  {10, 11},
  Wrap::Clamp,
};```

I'm looking to control it with a pot on the multiplexer, but its value needs to be calculated during the loop
such as with a 12 pole switch, with an increasing resistor on each pole
https://ae01.alicdn.com/kf/HTB11AIyQVXXXXa_XXXXq6xXFXXXN/RS26-1-Pole-Position-12-Selectable-1P12T-Band-Rotary-Channel-Selector-Switch.jpg_Q90.jpg_.webp

and on a side question, can I fetch values from the attached multiplexers in the loop without implimenting my own mux library?
I want to use some of the pins which aren't attached to notes for other things.

fuegovic, 2020-11-06:
About your side question, as far as I know you can fetch value from the mux in the loop with :

`mux.analogRead(pin)`

Here's an example where I print the value of mux pin 1 to the serial console and send the signal to usb to control modulation:

`
// Include the Control Surface library
#include <Control_Surface.h>
 
// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;
int muxPin1;

CD74HC4067 mux = {A3,{5, 4, 3, 2}
};

CCPotentiometer modulation = {
  muxPin1, {MIDI_CC::Modulation_Wheel, CHANNEL_12}
}; 

void setup() {
    Control_Surface.begin();
    Serial.begin(115200);
}
 
void loop() {
    Control_Surface.loop();
muxPin1 = mux.analogRead(1);
    Serial.println(muxPin1);
}`

tttapa, 2020-11-06:
You can implement your own selector, for example:

```cpp
template <setting_t N>
class PotSelector : public Updatable<> {
  public:
    PotSelector(Selectable<N> &selectable, pin_t analogPin) 
      : selectable(selectable), pot(analogPin) {}

    void begin() override {}
    
    void update() override {
      if (pot.update()) {
        setting_t setting = map(pot.getValue(), 0, 128, 0, N);
        selectable.select(setting);
      }
    }

  private:
    Selectable<N> &selectable;
    FilteredAnalog<7> pot;
};
```

For reading the input from a multiplexer, see https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d1b/1_8AnalogReadSerial_8ino-example.html.

digimbyte, 2020-11-07:
Thanks for the response, would this template be another file I create and import? 
I'm not too familair with arduino coding, my background is more node.js and c#

digimbyte, 2020-11-07:
here's my amature attempt at modifying the code
by passing the multiplexer directly and requesting the pin to read, should get better results - just not sure how to declare it.
```
template <setting_t N>
class PotSelector : public Updatable<> {
  public:
    PotSelector(Selectable<N> &selectable, CD74HC4067 _analogMux, int pinRead) 
      : selectable(selectable), pot(analogPin) {}

    void begin() override {}
    
    void update() override {
      if (pot.update()) {
        setting_t setting = map(_analogMux.analogRead(pinRead), 0, 128, 0, N);
        selectable.select(setting);
      }
    }

  private:
    Selectable<N> &selectable;
    FilteredAnalog<7> pot;
};
```

tttapa, 2020-11-08:
> by passing the multiplexer directly and requesting the pin to read, should get better results

Why? FilteredAnalog will read the mux input just fine.

```cpp
template <setting_t N>
class PotSelector : public Updatable<> {
  public:
    PotSelector(Selectable<N> &selectable, pin_t analogPin) 
      : selectable(selectable), pot(analogPin) {}

    void begin() override {}
    
    void update() override {
      if (pot.update()) {
        setting_t setting = map(pot.getValue(), 0, 128, 0, N);
        selectable.select(setting);
      }
    }

  private:
    Selectable<N> &selectable;
    FilteredAnalog<7> pot;
};

CD74HC4067 mux { A3, {5, 4, 3, 2} };
Transposer<-12, +12> transposer;
PotSelector<transposer.getNumberOfBanks()> selector { transposer, mux.pin(0) };
```

digimbyte, 2020-11-08:
AH, that makes so much more sense. I was expecting it to replace the transposer itself. 
Thank you. I would be lost otherwise!

closed 2020-11-08 by digimbyte