tttapa/Control-Surface #319: Global Integer as Value for CCPotentiometer is not working as it should....

fuegovic, 2020-11-06:
I have been trying to solve this issue for the past hours, it's probably a rookie mistake but I cannot find the solution.

I am trying to use a analog stick (like those on the xbox controller) for modulation and pitchbend. I got the pitchbend portion sorted but I have problems with the modulation.

I would like the stick to go from "full, to 0 in its resting position, to full". I worked out the algebra and I have the right numbers showing in the serial monitor. Good so far. 

The real issue is that when I try to use my global variable in the CCPotentiometer portion it completly bypass my calculation and use the original value of the potentiometer. So instead of being at 0 in the resting position I have half modulation.

At this point I am completly lost because the serial monitor show me something and the CCPotentiometer use another value and I have no idea where the other value is coming from. 

I hope I'm being clear, please help me! ;)

Thanks for your time!

```cpp
// regular pot:       0-----1023
//trying to do:    1023--0--1023
//               start-middle-end
//calculation are in void loop for now

// Include the Control Surface library
#include <Control_Surface.h>
 
// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;
int mod;
int val;
int mod2;

CD74HC4067 mux = {A3,{5, 4, 3, 2}
};

CCPotentiometer modulation = {
  mod2, {MIDI_CC::Modulation_Wheel, CHANNEL_12}
}; 

void setup() {
    Control_Surface.begin();
    Serial.begin(115200);
}
 
void loop() {
    Control_Surface.loop();
mod = mux.analogRead(11);
val = 1023 - mod;
  if (mod = 512)
  { 
    mod = 0;
  } 
  if (mod < 512)
  {
    mod = 1023 - (2 * mod);
  }
  if (mod > 512)
  {
    mod = 1023 - (2 * val);
  }

mod2 = abs(mod);

    Serial.println(mod2);
}

```

tttapa, 2020-11-06:
I think the main confusion is the arguments to the CCPotentiometer constructor. Please see the [documentation](https://tttapa.github.io/Control-Surface-doc/Doxygen/db/d32/classCCPotentiometer.html#acddf498ce3b8611d048f0f03f8c71eef) for it. As you can see, the first argument is the pin _number_, not the analog value. Secondly, the first argument is saved by value, so a copy of your `mod2` variable is made when initializing your `modulation` CCPotentiometer, so changing `mod2` in your loop would never affect the behavior of `modulation`.  

To map the values of a potentiometer, you have to use a mapping function, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d26/CCPotentiometer-Map_8ino-example.html.

Also, by writing `mod = 512`, you're assigning 512 to `mod`, it's not a comparison, you need `mod == 512` and then you shouldn't need the `abs(mod)` either.

```cpp
#include <Control_Surface.h>

USBDebugMIDI_Interface midi;

// regular pot:       0-----1023
// trying to do:    1023--0--1023
//                 start-middle-end
analog_t mapping(analog_t value) {
  analog_t maximum = FilteredAnalog<>::getMaxRawValue();
  analog_t halfway = maximum / 2;
  long centeredAroundZero = value - long(halfway);
  analog_t absoluteValue = std::abs(centeredAroundZero);
  analog_t mask = (halfway + 1) >> 1;
  analog_t highestBit = absoluteValue & mask;
  analog_t rounding = highestBit != 0;
  analog_t scaled = absoluteValue * 2 + rounding;
  return scaled;
}

CCPotentiometer potentiometer { A0, 7 };

void setup() {
  potentiometer.map(mapping);
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

fuegovic, 2020-11-06:
Wow, you are amazing, it works perfectly now! I will definitely have study the mapping documentation more. I did have the feeling I was on the wrong path but couldn't figure out what I should look for.

Thanks a million for your help!

closed 2020-11-06 by fuegovic