tttapa/Control-Surface #320: Error when compiling - "MIDIUSB.h: No such file or directory"

nage4200, 2020-11-07:
Arduino: 1.8.13 (Windows 7), Board: "Arduino Micro"

Hello,
I am a complete beginner with Arduino, and I am having an issue compiling my code for a MIDI controller that I am making.

-----------------------------------------------------------------------------------------------------------------------------------
**When I hit compile, I get this error message:**

In file included from C:\Users\(name)\Documents\Arduino\libraries\Control-Surface-master\src\MIDI_Interfaces\USBMIDI\USBMIDI.cpp:32:0:
C:\Users\(name)\Documents\Arduino\libraries\Control-Surface-master\src\MIDI_Interfaces\USBMIDI\USBMIDI_MIDIUSB.ipp:2:10: fatal error: MIDIUSB.h: No such file or directory
 #include <MIDIUSB.h>
          ^~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Micro.

I re-downloaded the library to make sure I didn't accidentally delete a file, but it still comes up with the same error.
------------------------------------------------------------------------------------------------------------------------------------

**Here is my code:**

```cpp
#include <Control_Surface.h>

HairlessMIDI_Interface midi;

Bank<4> bank(6);

IncrementSelectorLEDs<4> selector = {
  bank,
   {12} 
   {8, 9, 10, 11}
  Wrap::Wrap,
};

Bankable::PBPotentiometer generalUsePotentiometers[] = { 
  {{bank, BankType::CHANGE_CHANNEL}, A0, CHANNEL_1},
  {{bank, BankType::CHANGE_CHANNEL}, A1, CHANNEL_2},
};

Bankable::CCPotentiometer faders[] = { 
  {{bank, BankType::CHANGE_CHANNEL}, A3, {MIDI_CC::Channel_Volume, CHANNEL_3}},
  {{bank, BankType::CHANGE_CHANNEL}, A4, {MIDI_CC::Channel_Volume, CHANNEL_4}},
};

Bankable::NoteButtonLatching switches[] = {
  {{bank, BankType::CHANGE_ADDRESS}, 0, MCU::MUTE_1},
  {{bank, BankType::CHANGE_ADDRESS}, 1, MCU::MUTE_2},
};

CCPotentiometer mainVolumePot = {
  A2, {MIDI_CC::General_Purpose_Controller_1}
};

CCButton arcadebuttons[] = {
  {2, MIDI_CC::General_Purpose_Controller_2},
  {3, MIDI_CC::General_Purpose_Controller_3},
  {4, MIDI_CC::General_Purpose_Controller_4},
  {5, MIDI_CC::General_Purpose_Controller_5},
  {6, MIDI_CC::General_Purpose_Controller_6},
  {7, MIDI_CC::General_Purpose_Controller_7},
};

void setup() {
    Control_Surface.begin();
}

void loop() {
    Control_Surface.loop();
}

Any help would be greatly appreciated, as I am not exactly sure what to do at this point.

tttapa, 2020-11-07:
You forgot to install the MIDIUSB library. Please see the [installation instructions](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/da8/md_pages_Installation.html).

Also, don't use Hairless on an Arduino Micro, use a real USBMIDI_Interface.

nage4200, 2020-11-07:
Thank you so much for the quick reply!
Sorry for my dumb questions, but would I just switch out the "HairlessMIDI_Interface midi" with "USBMIDI_Interface"?
Thanks again!

closed 2020-11-07 by nage4200

tttapa, 2020-11-08:
> would I just switch out the "HairlessMIDI_Interface midi" with "USBMIDI_Interface"?

Yes, please see the [Getting Started page](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d7d/md_pages_Getting-Started.html#first-output-midi-interface) and the [MIDI Interfaces module](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/df0/group__MIDIInterfaces.html).

nage4200, 2020-11-08:
Ok, thanks for your help!