tttapa/Control-Surface #321: Help setting up MCU in logic pro X

mrmike12333, 2020-11-08:
Hi there,

I'm having trouble interfacing the control surface MCU with logic pro and I'm probably just not understanding how the framework is supposed to be used. 

I have a very simple script/setup where I want to control track faders in logic through 4 potentiometer sliders. The script is taken from a few code examples in the documentation (see bottom of this post).

When using the debug midi interface, I can see that the potentiometers are sending out information for the correct channel.
```
13:13:42.262 -> Control Change  	Channel: 2	Data 1: 0x00	Data 2: 0x18	Cable: 1
13:13:42.262 -> Control Change  	Channel: 2	Data 1: 0x00	Data 2: 0x17	Cable: 1
13:13:42.262 -> Control Change  	Channel: 2	Data 1: 0x00	Data 2: 0x15	Cable: 1
13:13:42.262 -> Control Change  	Channel: 2	Data 1: 0x00	Data 2: 0x13	Cable: 1
13:13:42.262 -> Control Change  	Channel: 2	Data 1: 0x00	Data 2: 0x12	Cable: 1
13:13:42.262 -> Control Change  	Channel: 2	Data 1: 0x00	Data 2: 0x11	Cable: 1
```

On the logic pro side, I'm not sure which Mackie device to use, however on trying all of them, changes in my potentiometers does not seem to change any of the volume faders in logic.

[image]

After choosing a mackie control surface, I can use and find the arduino on both the input and output port when configuring the control surfaces and the control surface shows in Logic that it's connected, however it does not work.

[image]

```cpp

#include <Control_Surface.h>

//USBMIDI_Interface Midi;
USBDebugMIDI_Interface midi(115200);

const uint8_t NumBanks = 1;
const uint8_t NumFaders = 4;

Bank<NumBanks> bank(NumFaders);

Bankable::CCPotentiometer faders[] = { 
  {bank, A0, MCU::VOLUME_1}, 
  {bank, A1, MCU::VOLUME_2}, 
  {bank, A2, MCU::VOLUME_3},
  {bank, A3, MCU::VOLUME_4}
  };

void setup() {
  RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
  Serial.begin(115200);
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

Any help or tips will be greatly appreciated!

tttapa, 2020-11-08:
The volume controls of the MCU protocol use Pitch Bend events, so you have to use `PBPotentiometer` instead of `CCPotentiometer`. You can see that the `MCU::VOLUME_1` constant belongs to the MCU Pitch Bend Controls group: https://tttapa.github.io/Control-Surface-doc/Doxygen/da/dc1/group__MCU__PB.html

mrmike12333, 2020-11-08:
Brilliant! It's working now, thanks for getting back so quickly

closed 2020-11-08 by mrmike12333