tttapa/Control-Surface #322: lock analog pin value on digital pin?

digimbyte, 2020-11-09:
One of the analog pins is a free spinning wheel that works like a potentiometer but it's value is based on its speed, this currently is connected to MIDI_CC breath controller.
I want to lock the value with when a mux digital pin is high, I need a custom pin wrapper? 
I am interested in having modifier from another mux digital pin: A) add to the value of the locked value, B) use highest value

Any idea's and help would be greatly appreciated!

```cpp
  { analogMux.pin(0), { MIDI_CC::Breath_Controller, CHANNEL_1 } }
```

tttapa, 2020-11-09:
As you might expect, this is not something that's supported out of the box. You'll have to program it yourself, either in the `loop` function or in a [custom MIDI Output Element](https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d3b/Custom-MIDI-Output-Element_8ino-example.html).

digimbyte, 2020-11-10:
Thanks, I was wondering if you could look at my code. I had some trouble with the documentation and C is still fairly foriegn. I don't know if I got the correct types for the analog pins since I was basing it on the output element as a start.

```
class FlyWheel: public MIDIOutputElement {
 public:
  FlyWheel(pin_t pin, pin_t cruise, pin_t mode, const MIDIAddress &address)
    : wheel{pin}, cruise{cruise}, cruisemode{mode}, address{address} {}
 
 public:
  // Initialize: enable the pull-up resistor for the button
  // This method is called once by `Control_Surface.begin()`.
  void begin() final override { 
    // wheel.begin(); // Analog doesn't appear to have this?
    cruise.begin();
    cruisemode.begin();
  }
 
  // Update: read the button and send MIDI messages when appropriate.
  // This method is called continuously by `Control_Surface.loop()`.
  void update() final override {
    uint8_t state_wheel = wheel.getValue();  // this is the analog pin to read
    AH::Button::State state_cruise = cruise.update(); // speed lock, aka cruise control
    AH::Button::State state_mode = cruisemode.update(); // mode, merge or highest
    
    if (state_cruise == AH::Button::Pressed) {
      if(state_mode == AH::Button::Pressed){ velocity = max(cruisespeed, state_wheel); }
      else { velocity = min(state_wheel + cruisespeed , 127); }
      Control_Surface.sendCC(address, velocity);  // → note on MIDIAddress  address, uint8_t   value 
    }
    else {
      Control_Surface.sendCC(address, velocity);  // → note on MIDIAddress  address, uint8_t   value 
    }
  }
 
 private:
  AH::FilteredAnalog<7> wheel;
  AH::Button cruise;
  AH::Button cruisemode;
  const MIDIAddress address;
  uint8_t cruisespeed;
  uint8_t velocity;
};

FlyWheel button = { analogMux.pin(0),digitalMux2.pin(9),digitalMux2.pin(10),   { MIDI_CC::Breath_Controller, CHANNEL_1 }, };
```

tttapa, 2020-11-10:
You never seem to call `wheel.update()`, so `wheel.getValue()` will always return the same value. See https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/dbe/1_8FilteredAnalog_8ino-example.html.

I can't comment on the logic for computing the velocity, that's something you'll have to debug yourself.

digimbyte, 2020-11-10:
Thanks! you are a life saver!

closed 2020-11-10 by digimbyte