tttapa/Control-Surface #323: How to change channel for every midi cc button in CCButtonMatrix?

kentforth, 2020-11-10:
I'm using CCButtonMatrix.
This skecth works fine for me, but all midi cc goes to CHANNEL_1

I would like to set different channels for every midi cc button
**How can I do that?**

I also would like to know how to use toggle buttons instead of momentary. I mean latched class for buttons.
**Can you give me  a sketch example?**

My Sketch:
```
#include <Control_Surface.h>

USBMIDI_Interface midi;

// The note numbers corresponding to the buttons in the matrix
const AddressMatrix<4, 4> addresses = {{
  { 0x24, 0x25, 0x26, 0x27},
  { 0x28,  0x29, 0x30, 0x31},
  { 0x32,  0x33, 0x34, 35},
  { 0x36,  0x37, 0x38, 0x39}
}};

CCButtonMatrix<4, 4> buttonmatrix = {
  {2, 3, 4, 5}, 
  {6, 7, 8, 9}, 
  addresses,    
    CHANNEL_1 
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

tttapa, 2020-11-10:
What you're asking is not supported out of the box, but you can easily implement it yourself, just create your own version of the `MIDIButtonMatrix` class:

https://github.com/tttapa/Control-Surface/blob/master/src/MIDI_Outputs/Abstract/MIDIButtonMatrix.hpp

Instead of using a matrix of addresses and single `baseChannelCN` member, use one matrix of MIDI addresses (controller number + channel), i.e. `Array2D<MIDIAddress, nb_rows, nb_cols>`.   
To make it latched, you'll have to keep track of the previous state, you could use a [`AH::BitArray`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/da4/classAH_1_1BitArray.html).

kentforth, 2020-11-13:
I'm not familiar with C++
What else should I create except class file to make it work in your library?

tttapa, 2020-11-14:
> I'm not familiar with C++

The Arduino platform and its libraries use C++ exclusively, so general C++ help is beyond the scope of this issues section.

You can create a class in your sketch, you don't have to create any extra files and you don't have to edit the library itself either. You just create your own class based on the `MIDIButtonMatrix` class I mentioned earlier. Then you probably also want to create a wrapper for ease of use, see  https://github.com/tttapa/Control-Surface/blob/master/src/MIDI_Outputs/NoteButtonMatrix.hpp. Finally, you create an instance of your class, and Control Surface will pick it up automatically.

kentforth, 2020-11-14:
Thanx a lot!

closed 2020-11-14 by kentforth