tttapa/Control-Surface #325: How to add midi button to this rgb code?

4dvn, 2020-11-11:
Hello Petter,
I have the code of rgb recived midi signal from Ableton and can change the color by the notes have velocity work like launchpad but i don't know how to add 16 midi buttons code and make it work together ( push the button and light up follow the rgb midi code )

```
/*
 *     USB MIDI to WS2812B by mat1jaczyyy & 4D
 *     ---------------------------------------
 */

/*  
 * LED Strip definition
 * --------------------
 */

const byte _NLED = 16;
const byte _DPIN = 2;

#include <Adafruit_NeoPixel.h>
Adafruit_NeoPixel _LED = Adafruit_NeoPixel(_NLED, _DPIN, NEO_GRB + NEO_KHZ800);

/*
 * Color Palette
 * Generate with retinaConverter.py (Retina 2.0+ Palette)
 * ------------------------------------------------------
 */

const byte _R[128] = {0, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 45, 93, 142, 223, 190, 28, 61, 93, 190, 125, 12, 28, 45, 158, 61, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 12, 28, 45, 158, 61, 28, 61, 93, 190, 125, 45, 93, 142, 223, 190, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 36, 73, 109, 146, 182, 219, 255};
const byte _G[128] = {0, 0, 0, 0, 125, 0, 12, 28, 45, 158, 61, 28, 61, 93, 190, 125, 45, 93, 142, 223, 190, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 45, 93, 142, 223, 190, 28, 61, 93, 190, 125, 12, 28, 45, 158, 61, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 36, 73, 109, 146, 182, 219, 255};
const byte _B[128] = {0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 12, 28, 45, 158, 61, 28, 61, 93, 190, 125, 45, 93, 142, 223, 190, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 45, 93, 142, 223, 190, 28, 61, 93, 190, 125, 12, 28, 45, 158, 61, 36, 73, 109, 146, 182, 219, 255};

/*
 * MIDI handler
 * ------------
 */

#include "MIDIUSB.h"
//#include "PitchToNote.h"

const byte _PStart = 36;  // First note in array
bool update = false;

void note(byte pitch, byte velocity) {
  _LED.setPixelColor(pitch - _PStart, _R[velocity], _G[velocity], _B[velocity]);
  update = true;
}

void setup() {
  _LED.begin();
  _LED.show();
}

void loop() {
  midiEventPacket_t rx;
  do { 
    rx = MidiUSB.read();
    if (rx.header == 0x9) note(rx.byte2, rx.byte3);
    if (rx.header == 0x8) note(rx.byte2, 0);
  } while (rx.header == 0);

  if (update) {
    _LED.show();
    update = false;
  }
}

//void sysEx(byte *data, unsigned int length) {
//  if (length == 6) {
//    _LED.setPixelColor(*(data+1) - _PStart, *(data+2), *(data+3), *(data+4));
//    update = true;
//  }
//}
```

tttapa, 2020-11-11:
You cannot use the MIDIUSB library when using Control Surface, you'll have to use Control Surface for everything.

The best solution is to use the `NoteRangeFastLED` class to drive your LED strip, see [Note-FastLED-ColorMapper.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d6d/10_8Note-FastLED-ColorMapper_8ino-example.html) and [Note-FastLED.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/dd2/9_8Note-FastLED_8ino-example.html). By default, it uses the Novation Launchpad color scheme, but you can provide your own color mapper as demonstrated in the example. 
The latest development version of Control Surface tells you when to update the LED strip as well, see [here](https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/d9/d6d/10_8Note-FastLED-ColorMapper_8ino-example.html).

If you want to re-use most of your existing code, you can use the MIDI input callbacks, see [MIDI-Input-Callback.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d32/MIDI-Input-Callback_8ino-example.html) or [MIDI-Input.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d1/d5d/MIDI-Input_8ino-example.html).

4dvn, 2020-11-11:
> re-use most of your existing code

I wanna re-use most of my existing code, can you give me the code of buttons with with my rgb code?

tttapa, 2020-11-11:
You already have working code with buttons in your [previous thread](https://github.com/tttapa/Control-Surface/issues/308). Have you checked the examples?

4dvn, 2020-11-11:
> You already have working code with buttons in your [previous thread](https://github.com/tttapa/Control-Surface/issues/308). Have you checked the examples? If you want help, please post specific questions, I won't be writing on-demand code.
Yes, is working but i want to try more with rgb, sorry for the rgb code, can you help me with that idea with controll surface library?

tttapa, 2020-11-11:
Instead of using `MidiUSB.read()` like in your first post, you use the Control Surface MIDI callbacks as shown here: https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d32/MIDI-Input-Callback_8ino-example.html

`cm.header` contains the MIDI status byte, if it's equal to `0x90`, you've received a Note On message on channel 1. If it's equal to `0x80`, it's a Note Off message on channel 1.  
`cm.data1` contains the note number (cfr. `rx.byte2`), and `cm.data2` contains the velocity (cfr. `rx.byte3`).

4dvn, 2020-11-11:
> Instead of using `MidiUSB.read()` like in your first post, you use the Control Surface MIDI callbacks as shown here: [...]

i'v stuck to made button work rgb 2 years :(  the idea i got from this video: https://www.youtube.com/watch?v=RMJNVMaHM7g

tttapa, 2020-11-11:
I can't help you if you don't tell me where you're stuck. What piece of the code are you having trouble with? 

The FastLED examples I linked to in my first reply just work. If you want to keep your RGB code (I wouldn't recommend it), you'll have a bit more work, as I explained in my previous posts.

tttapa, 2020-11-11:
Well, I gave you multiple options, have you tried them? What went wrong?

4dvn, 2020-11-11:
> Instead of using `MidiUSB.read()` like in your first post, you use the Control Surface MIDI callbacks as shown here: [...]

Can you upload the code of this maybe is hard to me to fix that code to keep my code and made it work with button code included on rgb code

tttapa, 2020-11-11:
> Can you upload the code of this maybe is hard to me to fix that code to keep my code and made it work with button code included on rgb code

I don't have the code of this, and I don't have time to write it. Please show some effort and try it yourself. The only difference is where the MIDI messages come from, either you get them from `MidiUSB.read()`, or the `channelMessageCallback()` gives them to you. If you get stuck, you can ask specific questions here.