tttapa/Control-Surface #327: Implement digital mux

dagame81, 2020-11-11:
[image]

Hi
I want to implement another mux but that is digital for push buttons.
Could you give me an example, I'm very new to this and I'm starting
thanks greetings
Please post the full code required to answer your question. Post the code even if it doesn't work. In that case, also include a description of what you want the code it to do.

dagame81, 2020-11-11:
I forgot to say that I want to implement a CD74HC4067 mux to have 2 one as analog like the one I show you in the screenshot and another digital one for push buttons

tttapa, 2020-11-11:
Please post the actual code, not pictures of code. The `CD74HC4067` class also supports digital inputs, you can use it for push buttons without any issues.

dagame81, 2020-11-11:
#include <Control_Surface.h> 

HairlessMIDI_Interface midi;

//MULTIPLEXOR POTENCIOMETROS
CD74HC4067 mux = {
  A0,
  { 2 ,3 ,4 ,5}
};
CCPotentiometer volumePotentiometer[] = {
  { mux.pin(0), {MIDI_CC::Channel_Volume, CHANNEL_1}},
  { mux.pin(1), {MIDI_CC::Channel_Volume, CHANNEL_2}},
  { mux.pin(2), {MIDI_CC::Channel_Volume, CHANNEL_3}},
  { mux.pin(3), {MIDI_CC::Channel_Volume, CHANNEL_4}},
  { mux.pin(4), {MIDI_CC::Channel_Volume, CHANNEL_5}},
  { mux.pin(5), {MIDI_CC::Channel_Volume, CHANNEL_6}},
  { mux.pin(6), {MIDI_CC::Channel_Volume, CHANNEL_7}},
  { mux.pin(7), {MIDI_CC::Channel_Volume, CHANNEL_8}},
  { mux.pin(8), {MIDI_CC::Channel_Volume, CHANNEL_9}},
  { mux.pin(9), {MIDI_CC::Channel_Volume, CHANNEL_10}},
  { mux.pin(10), {MIDI_CC::Channel_Volume, CHANNEL_11}},
  { mux.pin(11), {MIDI_CC::Channel_Volume, CHANNEL_12}},
  { mux.pin(12), {MIDI_CC::Channel_Volume, CHANNEL_13}},
  { mux.pin(13), {MIDI_CC::Channel_Volume, CHANNEL_14}},
  { mux.pin(14), {MIDI_CC::Channel_Volume, CHANNEL_15}},
  { mux.pin(15), {MIDI_CC::Channel_Volume, CHANNEL_16}},
};

//POTENCIOMETROS

CCPotentiometer potentiometers[] = {
  //{A0,        // Analog pin connected to potentiometer 1
   //0x10},     // Controller number of the first potentiometer
 //{A1,        // Analog pin connected to potentiometer 2
  // 0x11},     // Controller number of the second potentiometer
  //{A2, 0x12}, // Etc.
  //{A3, 0x13},
  //{A4, 0x14},
  //{A5, 0x15},
  //{A6, 0x16},
  //{A7, 0x17},
  //{A8, 0X18},
   //{A9, 0x19},
  //{A10, 0x20},
  //{A11, 0x21},
  //{A12, 0x22},
  //{A13, 0x23},
  //{A14, 0X24},
  //{A15, 0x25},
  
};

//PULSADORES

//Los pulsadores hay que cambiar el GENERAL PURPOSE CONTROLLER del 2 al 8 y cuando llegue al 8 cambiar el CHANNEL del 3 al 16 
// Pulsador 1 pin 53
CCButton button = {
  53,
  {MIDI_CC::General_Purpose_Controller_2, CHANNEL_2},
};
  // Pulsador 2 Pin 52
CCButton button1 = {
  52,
  {MIDI_CC::General_Purpose_Controller_3, CHANNEL_2},
};
// Pulsador 3 pin 51
CCButton button2 = {
  51,
  {MIDI_CC::General_Purpose_Controller_4, CHANNEL_2},
};
  // Pulsador 4 Pin 50
CCButton button3 = {
  50,
  {MIDI_CC::General_Purpose_Controller_5, CHANNEL_2},
};
// Pulsador 5 pin 49
CCButton button4 = {
  49,
  {MIDI_CC::General_Purpose_Controller_6, CHANNEL_2},
};
  // Pulsador 6 Pin 48
CCButton button5 = {
  48,
  {MIDI_CC::General_Purpose_Controller_7, CHANNEL_2},
};
// Pulsador 7 pin 47
CCButton button6 = {
  47,
  {MIDI_CC::General_Purpose_Controller_8, CHANNEL_2},
};
  // Pulsador 8 Pin 46
CCButton button7 = {
  46,
  {MIDI_CC::General_Purpose_Controller_2, CHANNEL_3},
};
// Pulsador 9 pin 45
CCButton button8 = {
  45,
  {MIDI_CC::General_Purpose_Controller_3, CHANNEL_3},
};
  // Pulsador 10 Pin 44
CCButton button9 = {
  44,
  {MIDI_CC::General_Purpose_Controller_4, CHANNEL_3},
};
// Pulsador 11 pin 43
CCButton button10 = {
  43,
  {MIDI_CC::General_Purpose_Controller_5, CHANNEL_3},
};
  // Pulsador 12 Pin 42
CCButton button11 = {
  42,
  {MIDI_CC::General_Purpose_Controller_6, CHANNEL_3},
};
// Pulsador 13 pin 41
CCButton button12 = {
  41,
  {MIDI_CC::General_Purpose_Controller_7, CHANNEL_3},
};
  // Pulsador 14 Pin 40
CCButton button13 = {
  40,
  {MIDI_CC::General_Purpose_Controller_8, CHANNEL_3},
};
// Pulsador 15 pin 39
CCButton button14 = {
  39,
  {MIDI_CC::General_Purpose_Controller_2, CHANNEL_4},
};
  // Pulsador 16 Pin 38
CCButton button15 = {
  38,
  {MIDI_CC::General_Purpose_Controller_3, CHANNEL_4},
};
// Pulsador 17 pin 37
CCButton button16 = {
  37,
  {MIDI_CC::General_Purpose_Controller_4, CHANNEL_4},
};
  // Pulsador 18 Pin 38
CCButton button17 = {
  38,
  {MIDI_CC::General_Purpose_Controller_5, CHANNEL_4},
};
// Pulsador 19 pin 37
CCButton button18 = {
  37,
  {MIDI_CC::General_Purpose_Controller_6, CHANNEL_4},
};
  // Pulsador 20 Pin 36
CCButton button19 = {
  36,
  {MIDI_CC::General_Purpose_Controller_7, CHANNEL_4},
};
// Pulsador 21 pin 35
CCButton button20 = {
  35,
  {MIDI_CC::General_Purpose_Controller_8, CHANNEL_4},
};
  // Pulsador 22 Pin 34
CCButton button21 = {
  34,
  {MIDI_CC::General_Purpose_Controller_2, CHANNEL_5},
};
// Pulsador 23 pin 33
CCButton button22 = {
  33,
  {MIDI_CC::General_Purpose_Controller_3, CHANNEL_5},
};
  // Pulsador 24 Pin 32
CCButton button23 = {
  32,
  {MIDI_CC::General_Purpose_Controller_4, CHANNEL_5},
};
// Pulsador 25 pin 31
CCButton button24 = {
  31,
  {MIDI_CC::General_Purpose_Controller_5, CHANNEL_5},
};
  // Pulsador 26 Pin 30
CCButton button25 = {
  30,
  {MIDI_CC::General_Purpose_Controller_6, CHANNEL_5},
};
// Pulsador 27 pin 29
CCButton button26 = {
  29,
  {MIDI_CC::General_Purpose_Controller_7, CHANNEL_5},
};
  // Pulsador 28 Pin 28
CCButton button27 = {
  28,
  {MIDI_CC::General_Purpose_Controller_8, CHANNEL_5},
};
void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

dagame81, 2020-11-11:
So it means that with this example I can connect the pushbuttons without problems without changing anything?

tttapa, 2020-11-11:
> So it means that with this example I can connect the pushbuttons without problems without changing anything?

Yes.

tttapa, 2020-11-11:
When you have to come ub with names like `button1`, `button2` etc., it almost certainly means you should be using arrays instead:

```cpp
CCButton buttons[] = {
  { 53, {MIDI_CC::General_Purpose_Controller_2, CHANNEL_2} },
  { 52, {MIDI_CC::General_Purpose_Controller_3, CHANNEL_2} },
  { 51, {MIDI_CC::General_Purpose_Controller_4, CHANNEL_2} },
  // ...
};
```

dagame81, 2020-11-12:
Thank you very much, I can already have 2 digital mux and another analog one, about the matrices for the pushbuttons I am very lost I do not know how to do them, whether or not to make matrices entails some problem or is it just to avoid having to write so much code?

dagame81, 2020-11-12:
i have 37 buttons i use a mega arduino board

tttapa, 2020-11-12:
Scanning button matrices are a way to read many buttons using a small number of input pins. They require more complicated hardware, because you have to create the matrix, and each switch needs a diode if you want to be able to press more than one button at once. They have some restrictions in software as well, they can only be used with the specific classes, like `NoteButtonMatrix`.

See https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d53/NoteButtonMatrix_8ino-example.html

If you already have the multiplexers, I'd use those instead of a scanning matrix.

dagame81, 2020-11-17:
thank you very much for your help