tttapa/Control-Surface #330: Change notes after setup?

digimbyte, 2020-11-14:
I have come across a situation where I need to swap a note with an entirely different one
if we can update it during runtime, that would be prefered but I would also be fine if the device has to restart
How? I could use some suggestions
` {transposer,  digitalMux2.pin(1), {note(F, 5), CHANNEL_1} },`
for
`
  {transposer,  digitalMux2.pin(1), {note(Gb, 5), CHANNEL_1} },`

tttapa, 2020-11-14:
This is impossible, by design.

Consider the following example: 

1. You create a NoteButton with address F5
2. You press the button
3. MIDI note F5 is turned on
4. You change the address to Gb5
5. You release the button
6. MIDI note Gb5 is turned off (!)

The result is that MIDI note F5 keeps playing indefinitely, and MIDI  note Gb5 is turned off, even though it wasn't playing in the first place.

Banks and Transposers avoid this situation, so they are the only way you're allowed to change the address of a MIDI element while the program is running.

digimbyte, 2020-11-14:
I'm aware of key ghosting, I thought there would be a function I could possibly call to close/turn off the note and assign a new one to that pin for it to swap around, even if its only updated after/on key release

transposers, do they shift the octave from F5 -> F4 or do they go down the notes A -> Bb -> B
etc

tttapa, 2020-11-14:
> I thought there would be a function I could possibly call to close/turn off the note and assign a new one to that pin for it to swap around, even if its only updated after/on key release

There isn't, because this would increase memory use and complicate the API. You could of course add it yourself, especially the `NoteButton` class is really simple, it's implemented in less than 10 lines of actual code, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d3b/Custom-MIDI-Output-Element_8ino-example.html.

> transposers, do they shift the octave from F5 -> F4 or do they go down the notes A -> Bb -> B
etc

By default, they transpose semitone by semitone, but you can specify any interval.

E.g. `Transposer<-12, +12> transposer` transposes semitone by semitone, up to a maximum of +12 semitones and a minimum of -12 semitones.  
`Transposer<-4, +4> transposer(12)` transposes octave by octave, up to a maximum of +4 octaves and a minimum of -4 octaves.

tttapa, 2020-11-14:
You can combine multiple transposers and other banks to change the addresses, you can find an example here: https://github.com/tttapa/Control-Surface/issues/317#issuecomment-722552940

It does require some boilerplate, as you can see.

digimbyte, 2020-11-18:
Thanks, I'll tinker with it and see how it goes

closed 2020-11-18 by digimbyte