tttapa/Control-Surface #332: Is it possible to use CD74HC4051 with rotary encoder ?

mhlsf, 2020-11-14:
**Post your code**
Please post the full code required to answer your question. Post the code even if it doesn't work. In that case, also include a description of what you want the code it to do.

```
#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
#include <Def/Channel.hpp>

 
// Instantiate a MIDI over USB interface.
//HairlessMIDI_Interface midi;
USBDebugMIDI_Interface midi;

Bank<2> bank(1);

// Instantiate an analog multiplexer
CD74HC4051 mux = {
  A0,       // Analog input pin
  {3, 4, 5} // Address pins S0, S1, S2
};

Button pushbutton = {2};

// Instantiate a CCRotaryEncoder object
MIDIAddress address = {12, CHANNEL_1, 0};

Bankable::CCAbsoluteEncoder<2> enc = {
  {bank, BankType::CHANGE_CHANNEL},
  {mux.pin(2), mux.pin(3)},       // pins
   address, // MIDI address (CC number + optional channel)
  6,            // optional multiplier if the control isn't fast enough
};

CCButton button = {
  // Push button on pin 5:
  mux.pin(0), 
  // General Purpose Controller #1 on MIDI channel 1:
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1},
};

 
void setup() {
  Control_Surface.begin(); // Initialize Control Surface
  Serial.print("start");
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

As title says, is it possible ? I tried with the code example overhead but it doesn't seem to work, the button work throught the multiplexer though.

Thanks

tttapa, 2020-11-15:
No, this is not possible, you can read the reasons behind this here: https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/df/df5/md_pages_FAQ.html#faq-mux-encoder

If you want to use a lot of encoders, the library only supports the MCP23017, if you want to use other means, you'll have to implement your own routines for reading the encoders. You just wrap it in a class with `void begin()` and `uint8_t read()` methods that initialize the encoder and return the absolute position, respectively, and then you can integrate it with the Control Surface MIDI rotary encoder classes, similar to the [MCP23017-RotaryEncoder-Interrupts.ino](https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/dd/d56/MCP23017-RotaryEncoder-Interrupts_8ino-example.html) example.

mhlsf, 2020-11-30:
Hi, I receive my MCP23017, in your link [MCP23017-RotaryEncoder-Interrupts.ino](https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/dd/d56/MCP23017-RotaryEncoder-Interrupts_8ino-example.html), you say that the MCP23017 only work on teensy board or arm devices, but I found this on [stackexchange](https://arduino.stackexchange.com/questions/52909/reading-several-rotary-encoders/52926#52926?newreg=bd2a68d35588440aaf7eadce997118e5), do you think your example code could also work with an arduino uno or I would be wasting my time trying ? Thank you !

tttapa, 2020-12-01:
The code in the StackExchange answer doesn't actually read the MCP23017 inputs inside of an interrupt, it's still done in the main loop. (https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/examples/Interrupt/InterruptExample.cpp)

There are two things that are referred to as “interrupts” here: 
1. The MCP23017 pulls down its interrupt pin when one of the inputs changes.
2. The microcontroller is interrupted by a changing microcontroller input pin.

The code from StackExchange uses both, but the second one is useless and the result is exactly the same when polling the MCP “interrupt” pin in the main loop. What it basically does is attaching a microcontroller external interrupt to the pin connected to the MCP interrupt pin. The handler for this interrupt just sets the `awakenByInterrupt` flag, and then the main loop polls that flag.  
It's easier and more efficient to poll the pin directly, using interrupts just to set a flag is pretty useless in most cases.

When truly using interrupts on a Teensy that has an I²C driver that can be used inside of an ISR, the MCP inputs are read inside of the interrupt handler instead of in the main loop.

You can use the [MCP23017-RotaryEncoder-No-Interrupts.ino](https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/d7/dac/MCP23017-RotaryEncoder-No-Interrupts_8ino-example.html) if you don't have a Teensy. It reads the MCP inputs in the main loop, just like the StackExchange code.

mhlsf, 2020-12-01:
I see, in terms of performance, would that be pretty noticeable is you only have 8 encoders ? I consider buying the teensy for the final project but until then I will try with the no interrupt, Thanks !

mhlsf, 2020-12-01:
Thanks for the precision !

closed 2020-12-01 by mhlsf