tttapa/Control-Surface #333: Trigger MIDI with a variable

AnnoyedSandwich, 2020-11-15:
Hey, 
is it possible to trigger a midi signal with a variable instead when a button is pressed?

For a little bit of context, I'm Working on a DMX to MIDI Converter. It listens for an incoming dmx signal on channel 1 and would then send a midi Note ON/OFF

I want to use this library for the MIDI part so my question is, is it possible to send a MIDI Signal when, for example, a boolean is true? So something like this 

`NoteButton example[] = {
(DMX Channel 1 = true), {note(C, 2), CHANNEL_1}
};`

tttapa, 2020-11-15:
Yes, you can send MIDI messages based on boolean values, but not using the `NoteButton` class, `NoteButton` is intended for buttons only.

To trigger the MIDI messages yourself, you can use the [`MIDI_Sender`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/d72/classMIDI__Sender.html) methods. `Control_Surface` implements these methods:

```cpp
#include <Control_Surface.h> // Include the Control Surface library
 
// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;
 
using namespace MIDI_Notes;

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
  
  if (/* DMX message received */) {
    uint8_t velocity = 127;
    Control_Surface.sendNoteOn({note(C, 2), CHANNEL_1}, velocity);
  } else if (/* other condition */) {
    uint8_t velocity = 127;
    Control_Surface.sendNoteOff({note(C, 2), CHANNEL_1}, velocity);
  }
}
```

If you don't need any of the MIDI controller-specific features of Control Surface, you can use the MIDI interface directly, instead of going through `Control_Surface`, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d8d/Send-MIDI-Notes_8ino-example.html and https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d9b/MIDI-Output_8ino-example.html.

AnnoyedSandwich, 2020-11-15:
That's awesome. Is there a similar class for control change events ?

tttapa, 2020-11-15:
Yes, if you go to the [`MIDI_Sender`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/d72/classMIDI__Sender.html) documentation page, you can see all methods, to send control change messages, you can use `Control_Surface.sendCC({ccnumber, channel}, ccvalue)`.