tttapa/Control-Surface #336: Invert buttons on multiplexer

digimbyte, 2020-11-17:
I've made some great progress with the suite, but after soldering parts together, I forgot you use a native pull-up on the buttons
but being a multiplexer, its hard to do.
```Bankable::NoteButton buttons[] = {
  {octaTransposer,  digitalMux1.pin(0), {note(Ab, 4), CHANNEL_1} },
  {octaTransposer,  digitalMux1.pin(1), {note(A,4), CHANNEL_1} },
  {octaTransposer,  digitalMux1.pin(2), {note(Bb, 4), CHANNEL_1} },
}
buttons.invert();
```

any thoughts and suggestions are welcome

digimbyte, 2020-11-18:
should I iterate through each button and invert them? 
I read in the documentation I can invert all buttons globally but no examples?

tttapa, 2020-11-19:
The `buttons` variable is an array, you cannot call methods on an array. You have to iterate and apply the method to each element of the array:

```cpp
void setup() {
  for (auto &button : buttons) // for each button in the buttons array
    button.invert();
  Control_Surface.begin();
}
```
It's not clear to me why you'd need to do this, though, Control Surface will use the internal pull-ups of the microcontroller, you don't need to add extra resistors, even when using multiplexers.

digimbyte, 2020-11-19:
I used a setup where buttons are connected on a main power rail with the potentiometers
I would have to resolder the button rails to ground to make it work otherwise?

tttapa, 2020-11-19:
The usual configuration is to wire all buttons between the Arduino input pin and ground, that's why there are internal pull-up resistors and usually no pull-down resistors.  

If you want to keep them connected to the positive power rail, you'll have to disable the internal pull-up resistor of the Arduino in your `setup` function, after calling `Control_Surface.begin()` and you'll have to provide your own (probably external) pull-down resistor(s).

digimbyte, 2020-11-19:
Thanks for the tip, Hope it all goes well!

closed 2020-11-19 by digimbyte