tttapa/Control-Surface #337: Rotary encoder problem

dagame81, 2020-11-17:
hello again today I have a problem with the rotary encoder, I post the code and an image of the hairless midi.
the problem is that when I want to map in traktor, the encoder does well in one direction but not in the other, that is:
I want to map the list of the browser and when turning the encoder to one side it does not go correlative, it makes ul jump to the beginning of the list but if I turn it the other way it does it well
[image]

#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
HairlessMIDI_Interface midi;

// Instantiate a CCRotaryEncoder object
CCRotaryEncoder enc = {
  {18, 19},       // pins
  MCU::V_POT_1, // MIDI address (CC number + optional channel)
  1,            // optional multiplier if the control isn't fast enough
};

void setup() {
  // Select the correct relative MIDI CC mode.
  // Options:
  //   - TWOS_COMPLEMENT (default)
  //   - BINARY_OFFSET
  //   - SIGN_MAGNITUDE
  //   - NEXT_ADDRESS
  // Aliases:
  //   - REAPER_RELATIVE_1
  //   - REAPER_RELATIVE_2
  //   - REAPER_RELATIVE_3
  //   - TRACKTION_RELATIVE
  //   - MACKIE_CONTROL_RELATIVE
  //   - KORG_KONTROL_INC_DEC_1
  RelativeCCSender::setMode(relativeCCmode::MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

dagame81, 2020-11-17:
in the direction that works well the value is 65,
in the address that works badly gives the value 1

tttapa, 2020-11-17:
This is the expected behavior. If the mode you have selected (`MACKIE_CONTROL_RELATIVE`) with Traktor, you'll have to use a different mode, either change your settings in Traktor, or use the correct mode in your Arduino code, try `TWOS_COMPLEMENT`, for example.

dagame81, 2020-11-19:
Hello again
I've been trying all the modes that you incorporate and in this mode it works well for the browser, but not for the jog whell to adjust the tempo bend, when turning the encoder it is always activating the tempo bend function and it does not stop, to stop it I have than turn the encoder to the other direction one pulse.
I would like that every time I turn the encoder it activates the tempo bend but it does not stay activated continuously

#include <Encoder.h> 
#include <Control_Surface.h> 

HairlessMIDI_Interface midi;

CCRotaryEncoder enc = {
  {18, 19},       // pins
  MCU::V_POT_1, // MIDI address (CC number + optional channel)
  1,            // optional multiplier if the control isn't fast enough
};

void setup() {

  RelativeCCSender::setMode(relativeCCmode::BINARY_OFFSET);

  
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop(); 
}

dagame81, 2020-11-19:
I have tried to configure the traktor in every way but I have the same problem again

tttapa, 2020-11-19:
I don't use Traktor myself, but AFAIK, you have to select the relative 7Fh/01h mode in the settings, and then use `TWOS_COMPLEMENT` on the Arduino:
[image]

Control Surface supports all relative modes I'm aware of, but I don't know how Traktor handles them. If a mode works for some controls in Traktor but not for others, then that would be a question to ask on a Traktor forum or to look up in the Traktor documentation or MIDI implementation. For reference, here's the documentation of the different encoding modes supported by Control Surface: https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d24/RelativeCCSender_8hpp.html#a793b4d05f50a2c3af3ddeeef50e7a71d

dagame81, 2020-11-19:
Works fine on everything except tempo bend when using encoder.
When turning a single beat, it activates the tempo bend and does not stop until I turn one beat in the other direction, if I turn more beats it becomes active again without stopping.
If there is no solution I will have to look for other alternatives.
Could you introduce another code that can search the network to see if traktor uses other protocols?

dagame81, 2020-11-19:
[image]

found this information of the encoders that I bought, could it be that I need a 5 v resistors on pins A and B?

dagame81, 2020-11-19:
resistor connected to 5 volts I meant

tttapa, 2020-11-19:
> Could you introduce another code that can search the network to see if traktor uses other protocols?

What do you mean? MIDI is not a network, you cannot search it, the Arduino cannot detect if it's talking to Traktor or not.

> found this information of the encoders that I bought, could it be that I need a 5 v resistors on pins A and B?

The pull-up resistors are internal to the microcontroller pins, you don't have to add any external ones.

dagame81, 2020-11-19:
Been testing the encoders of a traktor controller that is the X1 and it does the same, therefore it is not a problem of code or wiring, it is seen that traktor uses another way to map that function and I imagine that it would have to update to the version scratch to work well

dagame81, 2020-11-19:
Thank you very much for your attention and great help, I am very close to finishing my project, with the help that you provide, everything becomes easier.
When I finish my project I invite him to a beer, I think he is far from me but I will try ahahahah
Thank you so much

dagame81, 2020-11-19:
I still have some small doubts that I will open comments but the most difficult thing I already have

closed 2020-11-19 by dagame81