tttapa/Control-Surface #338: Turn on led with button

dagame81, 2020-11-22:
Hello again,
I would like that every time I press a button an LED would light up and stay on.
I don't understand much of this but I've been testing the examples you include and I don't know how to do it, I need 4 pushbuttons and 4 LEDs
Thank you

dagame81, 2020-11-24:
#include <Control_Surface.h>
HairlessMIDI_Interface midi;

using namespace MIDI_Notes;
NoteValueLED noteLed = { 13, {note(C,4), CHANNEL_1}};
CCButton button = {
  // Push button on pin 5:
  7,
  // General Purpose Controller #1 on MIDI channel 1:
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1},
};

void setup() {
 Control_Surface.begin();

}

void loop() {
  Control_Surface.loop();
}

dagame81, 2020-11-24:
I am testing with this code, but the led does not turn on neither when I map in traktor

dagame81, 2020-11-24:
#include <Control_Surface.h>

HairlessMIDI_Interface midi;
using namespace MIDI_Notes;

CCButton button = {7, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_5}};

constexpr pin_t ledPin = 12;
NoteValueLED led = {
  12,             // Pin of built-in LED
  {note(C, 4), CHANNEL_1}, // Note C4 on MIDI channel 1
};

void setup() {
  Control_Surface.begin();
  pinMode(ledPin, OUTPUT);
}

void loop() {
  Control_Surface.loop();
  if (button.getButtonState() == Button::Rising)
    digitalWrite(ledPin, LOW);
  else if(button.getButtonState() == Button::Falling)
    digitalWrite(ledPin, HIGH);
}

I'm testing with this code and the led already turns on but I can't map it in traktor.
What am I doing wrong?

tttapa, 2020-11-24:
> ```c++
> NoteValueLED noteLed = { 13, {note(C,4), CHANNEL_1}};
> CCButton button = { 7, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}};
> ```

The first problem is that the LED listens for MIDI note messages for note C4, but your button sends Control Change messages to General Purpose Controller 1. They are not related whatsoever, so the LED will not react to button presses in any way.

Second, the `NoteValueLED` class listens for _incoming_ MIDI messages, while the `CCButton` class sends _outgoing_ MIDI messages. This means that they can only interact if the software/hardware at the other end of the MIDI connection somehow loops back these messages, which is often not the case (at least not by default).

`NoteValueLED` is intended to listen for feedback from the DAW, it doesn't depend on the state of your buttons.

> I'm testing with this code and the led already turns on but I can't map it in traktor.
> What am I doing wrong?

What are you trying to map and what doesn't work?

If you want the LED to depend only on the button state, you have to use the `CCButton::getButtonState()` method, like you did in your last example, but you shouldn't be using `NoteValueLED` in that case:

```cpp
#include <Control_Surface.h>

HairlessMIDI_Interface midi;

CCButton button = {7, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_5}};

constexpr pin_t ledPin = 12;

void setup() {
  Control_Surface.begin();
  pinMode(ledPin, OUTPUT);
}

void loop() {
  Control_Surface.loop();
  if (button.getButtonState() == Button::Rising)
    digitalWrite(ledPin, LOW);
  else if(button.getButtonState() == Button::Falling)
    digitalWrite(ledPin, HIGH);
}
```
You should be able to map the CCButton in Traktor without any issues.

dagame81, 2020-11-25:
What does not allow me to map is the led, the button maps it perfectly but the led does not.
In traktor as in other softwares I suppose that you have to assign a CC channel or Note to the led
[image]

dagame81, 2020-11-25:
I have tried it with the examples that you have especially the NOTE-LED example assigned to it the C-4 but nothing.

#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
HairlessMIDI_Interface midi;

using namespace MIDI_Notes;

// Instantiate the LED that will light up when middle C is playing
NoteValueLED led = {
  LED_BUILTIN,             // Pin of built-in LED
  {note(C, 4), CHANNEL_1}, // Note C4 on MIDI channel 1
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
[image]

tttapa, 2020-11-27:
Your image shows a mapping on channel 3, while your code uses channel 1.

I'm not familiar enough with Traktor to really help you, but my guess is that it's either not mapped correctly, or that Traktor doesn't send MIDI output data/feedback for LEDs by default. If Traktor is configured correctly, the outgoing MIDI messages should show up in the Hairless debug log. Do you see them?

dagame81, 2020-11-30:
Hello again ,
the problem is traktor that does not send data to hairless.
I already solved it, when using mac, I had to create a separate midi device so that traktor can send the output records
Thank you very much for your work, I am learning a lot with you
Greetings Pieter