tttapa/Control-Surface #339: How to use program change with banks?

kentforth, 2020-11-23:
Let's say I have 2 buttons for sending program change and 2 buttons for changing banks("btnUp", "btnDown")
Initially  program change buttons send `MIDI_PC::Acoustic_Grand_Piano` and `MIDI_PC::Bright_Acoustic_Piano`
I I press btnUp button the first PC button chanegs to `MIDI_PC::Electric_Grand_Piano` and second PC button changes to `MIDI_PC::Gunshot`

If I press btnDown button these buttons go to it's inital PC value

**How can I do it? Can you gvie me a sample sketch?**

There is program change example in your library, but I don't know how to use it with banks:
```
ProgramChanger<4> programChanger = {
  {
    MIDI_PC::Acoustic_Grand_Piano, // list of programs
    MIDI_PC::Bright_Acoustic_Piano,  
  },
  CHANNEL_1, // MIDI channel to use
};
```

tttapa, 2020-11-27:
It's not clear to me what you mean, ProgramChanger doesn't use banks.

If you want to control it using two buttons, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d7f/Program-Changer_8ino-example.html and use the [IncrementDecrementSelector](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/deb/classIncrementDecrementSelector.html) instead of the ManyButtonsSelector.

kentforth, 2020-11-27:
it is not necessary for me to use only programChanger. I just need to change program change number of buttons with banks

kentforth, 2020-11-27:
Thank you for answer!

closed 2020-11-27 by kentforth

tttapa, 2020-11-27:
> it is not necessary for me to use only programChanger. I just need to change program change number of buttons with banks

In that case, you might want to use the [`Bankable::PCButton`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/d5c/classBankable_1_1PCButton.html) class instead of `ProgramChanger`.