tttapa/Control-Surface #340: Add "seven segment led display"

santta, 2020-11-23:
hello, Pieter, how add "seven segment led display" to indicate the number "IncrementDecrementSelector"?
[image]
[image]

tttapa, 2020-11-27:
This is currently not supported by Control Surface, but it's easy enough to program yourself, independently from Control Surface. You can get the current selection from the selector using the `IncrementDecrementSelector::get()` method.

santta, 2020-11-29:
Thanks for the answer, I will try it, then I will definitely share the sketch.

closed 2020-11-29 by santta