tttapa/Control-Surface #342: 74HC151 Digital mux for buttons instead of analog mux

Guidosmus, 2020-11-26:
Hello,
First of all many thanks for making this awesome library available! It already gave me some really nice arduino/teensy experience without any prior experience.

My question is if the SN74HC151 digital multiplexer is in anyway supported for reading pushbuttons? I've tried (ab)using the analogmux support to get it working but I don't think it works that easy because it expects to be reading analog values? See some commented out parts for my other newbish experiments.

Excuse my inexperience.

Kind regards,

Guido
```cpp
// Your code here
// Include the library
#include <Control_Surface.h>
 
// Instantiate a MIDI Interface to use
USBMIDI_Interface midi;

//Tried this in the hope an actual "class" for this mux was available but it is not
// Instantiate an "digital" multiplexer
//SN74HC151N mux = {
//  11,       // digital input pin
//  {3, 4, 5} // Address pins A, B, C
//};

//Tried to abuse analog mux "class" without any luck
// Instantiate an "digital" multiplexer
CD74HC4051 mux = {
  11,       // digital input pin
  {3, 4, 5} // Address pins A, B, C
};

//This is the example
// Instantiate an analog multiplexer
//CD74HC4051 mux = {
//  11,       // Analog input pin
//  {3, 4, 5} // Address pins S0, S1, S2
//};
 
// Create an array of potentiometers that send out
// MIDI Control Change messages when you turn the
// potentiometers connected to the eight input pins of
// the multiplexer
CCButton buttons[] = {
  {mux.pin(0), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}},
  {mux.pin(1), {MIDI_CC::General_Purpose_Controller_2, CHANNEL_1}},
  {mux.pin(2), {MIDI_CC::General_Purpose_Controller_3, CHANNEL_1}},
  {mux.pin(3), {MIDI_CC::General_Purpose_Controller_4, CHANNEL_1}},
  {mux.pin(4), {MIDI_CC::General_Purpose_Controller_5, CHANNEL_1}},
  {mux.pin(5), {MIDI_CC::General_Purpose_Controller_6, CHANNEL_1}},
  {mux.pin(6), {MIDI_CC::General_Purpose_Controller_7, CHANNEL_1}},
  {mux.pin(7), {MIDI_CC::General_Purpose_Controller_8, CHANNEL_1}},
};
 
// Initialize the Control Surface
void setup() {
  Control_Surface.begin();
}
 
// Update the Control Surface
void loop() {
  Control_Surface.loop();
}
```

tttapa, 2020-11-27:
You should be able to use the `AnalogMultiplex` class (and `CD74HC4051`) with your digital multiplexer. However, you will need a pull-up resistor on each of the 8 input pins of the multiplexer if you're working with buttons.

Guidosmus, 2020-12-10:
Many thanks for your response! I got it working eventually. My newbish assumption that the one pullup resistor would work for all the pins of the multiplexer was very wrong. And beside that I accidentally connected a inverted pin of the mux to ground. After fixing these two things it works like a charm! Thank you very much once more!

Since I plan to have approximately 30 buttons and 30 pots(these of course will use analog muxes) in total, is there any benefit to using the CCButton**s** and it's CCPotentiometer**s**(if it exists)?

And if so how would one go about and manage the available CC event addresses? Maybe this should not be in this same thread, if so I understand.

tttapa, 2020-12-15:
> Since I plan to have approximately 30 buttons and 30 pots(these of course will use analog muxes) in total, is there any benefit to using the CCButtons and it's CCPotentiometers(if it exists)?

CCButtons and CCPotentiometers group multiple buttons/potentiometers in a single object, rather than having one CCButton/CCPotentiometer object for each button/pot. This reduces the memory overhead, and is a tiny bit faster. I haven't done any performance tests, so I don't know how large the benefits will be.  
Another advantage might be that CCButtons/CCPotentiometers can be initialized more compactly, since you don't have to specify each address individually, you simply provide a starting address and the offset between successive addresses, the address for each control is computed from these two values automatically.

> And if so how would one go about and manage the available CC event addresses?

I'm not sure what you mean by this?

Guidosmus, 2020-12-21:
And again I really appreciate you taking the time to answer questions! I understand both benefits of using CCButtons and CCPotentiometers thanks for the clear explanation and I will trust you on that it's more efficient. The part where I run into trouble is understanding how to manage what MIDI addresses will be chosen.I think I understand the parameters you described for computing the addresses. What I don't understand is how I will not run into trouble when I create multiple object all needing midi adresses.

What I understand from MIDI is that it has a limited address space per channel. Of the available addresses quite a lot seem to be reserved for specific functions(which I assume should be avoided when not used for their specific purpose?). So if I understand correctly the total amount of "undefined CC controllers" per channel is 37. Is the computing of the addresses taking this guideline into account or is there actually no need for this and is my assumption just incorrect? e.g. one can just start at address "1" and count up until all addresses are used regardless of there prepurposed function?

I also understand that every address is theoretically the same and in case of buttons values will only be 0 or 127 but for knobs etc it will be 1-127.

Sorry for the bad explanation but it's hard to ask a defined question when lost in confusion. Could be that my MIDI knowledge is just lacking.

tttapa, 2020-12-21:
MIDI has two types of addresses: a 4-bit channel and a 7-bit address within a channel. There are 2⁷ = 128 addresses on each of the 2⁴ = 16 channels. Each message type (note or control change) has its own address space.

The MIDI standard assigns meaning to most controller addresses, e.g. address 7 is “channel volume”. Some synthesizers might care about these assignments, but most DAWs don't, so for a MIDI DAW controller, you can use any address from 0-119 without any issues.

Control Change addresses 120-127 are not actually controllers but have special meanings (special commands to turn off all playing notes, for instance), so you should avoid them. That leaves you with 16×120 = 1920 controllers in total, which should be more than enough.

For MIDI notes, you can use any address from 0-127.

Guidosmus, 2020-12-23:
Many thanks for that quick explanation. I'll close this question(s) since it's already spanning multiple topics and your answers are in the earlier comments! 

Enjoy the upcoming days.

closed 2020-12-23 by Guidosmus