tttapa/Control-Surface #343: reset transposer

benwadub, 2020-11-27:
**Post your code**
hi citer, hope you are fine? I just seen in an exemple that pushing transposer up and down at the same time will reset the transposer, how could I reset my encoder transposer when pushing on the push button of the encoder that is wired to pin 3 please?

```cpp
CCButtonLatched button1 = {3, {120, CHANNEL_16}};
// Instantiate an encoder to change the transposition
EncoderSelector<transposer.getNumberOfBanks()> selector = {
  transposer,
  {4, 5},
  4,
  Wrap::Wrap,
};
```

tttapa, 2020-11-27:
If you look at the documentation for the [`EncoderSelector`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/dc5/classEncoderSelector.html#a494a9045acc596fbdcacdd56c01f8950) constructor, you'll see that the second argument is of type [`EncoderSwitchPinList`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/dc2/structEncoderSwitchPinList.html), which has the `switchPin` as a third optional constructor parameter.

```cpp
EncoderSelector<transposer.getNumberOfBanks()> selector = {
  transposer,
  {4, 5, 3},
  4,
  Wrap::Wrap,
};
```

benwadub, 2020-11-28:
You mean that if I write like this my button will automatically become a reset button for the encoder selector?