tttapa/Control-Surface #344: MUX "A,B,C" pins on Analog Inputs

sucrz, 2020-11-29:
Hello, 

I tried to use a multiplexer (CD4051BE) to read buttons, the issue is the only pins on my arduino nano board available for mux "ABC" pins are A5,A6,A7. 
I've tried connect ABC pin to digital pins and it works fine but for my "final version" I have to use pins A5, A6, A7. 

For what I undestand the issue is the use of AnalogRead/Write in the "AnalogMultiplex.hpp" file.
Since I'll only connect buttons to mux, can you point me to the lines to change in order to connect A5,A6, A7 to A,B,C mux pins. 

Thanks

tttapa, 2020-12-08:
Pins A6 and A7 are not digital pins on an Arduino Nano, they can only be used for analog input.

For the address lines of the multiplexer, you can use any digital pin [0-13] or any analog pin [A0-A5] but not A6 or A7.

Normanras, 2021-02-22:
@sucrz Would you mind sharing your code? I just posted to the discussion a question with the CD4051BE, and my teensy/control surface wasn't reading it correctly. Granted, I'm using faders on the mux instead of buttons, but if your code works, I may be able to glean something from it!