tttapa/Control-Surface #345: Pulsaciones simultaneas maxima en una matrix

alomdj00, 2020-11-30:
Hola, basicamente tengo dos preguntas.
1. Cuantos botones se pueden presionar al mismo tiempo en una matrix?

2. Si tengo dos matrices o mas, puedo usar los mismos pines de rows o de cols para ahorrar uno pines, o los pines tienen que ser diferentes entre la matrix1 y la matrix2.

Muchas gracias

**Post your code**
Please post the full code required to answer your question. Post the code even if it doesn't work. In that case, also include a description of what you want the code it to do.

```cpp
// Your code here
```

tttapa, 2020-11-30:
1. If you have diodes in your matrix, as shown in the [schematic in the example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d53/NoteButtonMatrix_8ino-example.html), you can press all buttons at once. Without diodes, you can only press one or two buttons at the same time.
2. Yes, you can reuse either the column pins or the row pins, but not both. You're essentially just creating a larger matrix.

alomdj00, 2020-11-30:
Puedo utilizar multiplexores para las entradas de la matriz, las columnas

tttapa, 2020-11-30:
Yes, but not for the outputs.