tttapa/Control-Surface #346: Single Pin, Multiple MIDI Mappings

Normanras, 2020-11-30:
Hi,

I've been testing lots of different MIDI Arduino/Teensy codes today and so far, with the Control Surface Library, yours is the best and most comprehensive. Best of all - it works! Your explanation and documentation is just stellar. Thanks for all your hard work. When this controller is completed, I'll do a write up and heavily thank/reference you!

I am still on the "Multiple Control Change Potentiometers" section of the files and have a question. I am testing a few sliding potentiometers with a few different synth VSTs, right now with the open-source "Surge Synth". The synth allows you to right click a slider/fader and choose "MIDI Learn Parameter" and then move the potentiometer and it pairs. If I then select another slider on the VST and move the same potentiometer, it connects to both. So one potentiometer moves both digital sliders. 

Is there any commands i can give the arduino to only connect to one slider at a time? Ideally, if the pot is connected to slider a, and then on slider b I select MIDI Learn Parameter and move that same pot, it will automatically pair with slider b and unpair from slider a. 

Thanks again for all your hard work! I will reply here if I find the code to work with that, but I figured I would ask the creator themself.

tttapa, 2020-12-01:
MIDI mapping/learning is implemented in the receiving software entirely, so unfortunately there's nothing you can do on the Arduino to prevent it from mapping to the same control twice.

Normanras, 2020-12-01:
Learned something new. Didn't know that - thanks!

tttapa, 2020-12-01:
MIDI controllers usually just sends messages like `Controller #7: Value 45%`. It's up to the software on the other side to interpret that message and take actions based on the controller number and/or value.

MIDI learn mode just associates the MIDI controller numbers with sliders and knobs in the DAW or plugin.

nbazzeghin, 2021-02-21:
@Normanras The reason you are getting both mapped to the is likely you are sending the same CC value on different channels. You can fix this by changing the CC to be different and use the same channel.

Here is an example I am using with a mux chip, but you could just replace mux.pin w/ A0, A1, etc.:
```
CCPotentiometer volumePotentiometers[] = {
  {mux.pin(0), {102, CHANNEL_10}},
  {mux.pin(1), {103, CHANNEL_10}},
  {mux.pin(2), {104, CHANNEL_10}},
  {mux.pin(3), {105, CHANNEL_10}},
  {mux.pin(4), {106, CHANNEL_10}},
  {mux.pin(5), {107, CHANNEL_10}},
  {mux.pin(6), {108, CHANNEL_10}},
  {mux.pin(7), {109, CHANNEL_10}},
  {mux2.pin(0), {110, CHANNEL_10}},
  {mux2.pin(1), {111, CHANNEL_10}},
  {mux2.pin(2), {112, CHANNEL_10}},
  {mux2.pin(3), {113, CHANNEL_10}},
  {mux2.pin(4), {114, CHANNEL_10}},
  {mux2.pin(5), {115, CHANNEL_10}},
  {mux2.pin(6), {116, CHANNEL_10}},
  {mux2.pin(7), {117, CHANNEL_10}},
};
```