tttapa/Control-Surface #347: PC Button array vs Program Changer

kentforth, 2020-12-01:
Pieter, please tell me what is difference between PCButton array of buttons and Program Changer

If I declare buttons like this:

```
ProgramChanger<2> programChanger = {
  {
    22, // list of programs
    45,    
  },
  CHANNEL_1, // MIDI channel to use
};

// Instantiate a selector that reads three buttons and controls the program
// changer
ManyButtonsSelector<2> programSelector = {
  programChanger,
  {{2, 3}},
};

//button array of program change
PCButton buttons[] = {
  {2, {22, CHANNEL_1}}, //button on pin 2 send MIDI CC 46
  {3, {45, CHANNEL_1}}, 
 };
```

Both methods of using program change leads to the same result. Buttons changes PC number

**What is difference?**

tttapa, 2020-12-01:
If you use a `ManyButtonsSelector`, they are pretty much the same. The main difference is that you can use any [`Selector`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d0c/group__Selectors.html) with the `ProgramChanger` class, while the `PCButton` class just sends a program change message when a button is pressed.  
`PCButton` also has a Bankable version if you need to change the channel or program numbers.

kentforth, 2020-12-01:
Now it is clear enough for me, Thankx!

closed 2020-12-01 by kentforth