tttapa/Control-Surface #348: potentometer mapping on mux

timce2000, 2020-12-03:
hi is it possible to map linear potentiometers connected to analog multiplexer CD74HC4067 i tried using the same options as for analog pin bit i get the following error
```cpp
software:34:10: error: request for member 'map' in 'faders', which is of non-class type 'CS::CCPotentiometer [6]'
   faders.map(mappingFunction);
          ^~~
exit status 1
request for member 'map' in 'faders', which is of non-class type 'CS::CCPotentiometer [6]'
```
i would like to map them because potentiometer 1 jumps from midi channel 1 and 2 and the second potentiometer isnt working and potentiometer 3 jumps from ch 3 to 4 and potentimeter 4 doesnt work

```cpp
#include <Control_Surface.h>
#include <MIDI_Interfaces/SerialMIDI_Interface.hpp>

USBMIDI_Interface midi;

CD74HC4067 mux = {
  A0, {8, 9, 10, 11} 
};

CCPotentiometer faders[] = {
  { mux.pin(0), { MIDI_CC::Channel_Volume, CHANNEL_1 } },
  { mux.pin(1), { MIDI_CC::Channel_Volume, CHANNEL_2 } },
  { mux.pin(2), { MIDI_CC::Channel_Volume, CHANNEL_3 } },
  { mux.pin(3), { MIDI_CC::Channel_Volume, CHANNEL_4 } },
  { mux.pin(4), { MIDI_CC::Channel_Volume, CHANNEL_5 } },
  { mux.pin(5), { MIDI_CC::Channel_Volume, CHANNEL_6 } },
};

//Potentiometer mapping
constexpr analog_t minimumValue = 255;
constexpr analog_t maximumValue = 16383 - 255;
analog_t mappingFunction(analog_t raw) {
  raw = constrain(raw, minimumValue, maximumValue);
  return map(raw, minimumValue, maximumValue, 0, 16383);
}

void setup() {
  faders.map(mappingFunction);
  Control_Surface.begin();

}

void loop() {
  Control_Surface.loop();
}
```

tttapa, 2020-12-03:
`faders` is an array, not an object, you cannot invoke methods against an array. If you want to apply the mapping function to each object in the array, use a for loop:

```cpp
for (auto &fader : faders)
  fader.map(mappingFunction);
```

> i would like to map them because potentiometer 1 jumps from midi channel 1 and 2 and the second potentiometer isnt working and potentiometer 3 jumps from ch 3 to 4 and potentimeter 4 doesnt work

I don't see how using a mapping function would solve any of that. It sounds like there's something wrong with the wiring of your multiplexer.

timce2000, 2020-12-03:
ok thanks i will check the wiring but by looking at the wiring with naked eye there arent any problems