tttapa/Control-Surface #349: Is there a simple way for a regular button to "unlatch" another button? (stop btn , rec latched btn)

fuegovic, 2020-12-03:
I had to use a latched record button because of the way my stuff is setup in ableton live with my max for live plugin and I was wondering if there is a simple way for my stop button to also change the state of my record button to "0" or "off".

```cpp
CCButton buttons[] = {
  {14, {MIDI_CC::General_Purpose_Controller_2, CHANNEL_1}}, // STOP
  {16, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}}, // PLAY
};
CCButtonLatched LatchedButtons[] = { 
  {15, {MIDI_CC::General_Purpose_Controller_3, CHANNEL_1}}, // REC
};
```
I tried to send a custom midi message from the loop but that obviously did not work. The state is not turned off in the library so I still have to press the record button twice to re-enable it...

```cpp
 if (digitalRead(14) == LOW){
      Control_Surface.sendCC(MIDI_CC::General_Purpose_Controller_3, 0); 
 }
```

I was looking for a "simple" way for my stop button to also change the state of the record button so I don't need to press it twice the next time I use it.

tttapa, 2020-12-03:
You can use [`CCButton::getButtonState()`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/d72/classMIDIButton.html#a2c8c834002cd1700bb1da13a96935dd1) and [`CCButtonLatched::setState()`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d88/classMIDIButtonLatched.html#ac283492d8697643beb987ac446bd3fbb):

```cpp
CCButton stop {14, {MIDI_CC::General_Purpose_Controller_2, CHANNEL_1}};
CCButtonLatched rec {15, {MIDI_CC::General_Purpose_Controller_3, CHANNEL_1}};
...
if (stop.getButtonState() == Button::FALLING)
  rec.setState(false);
```

fuegovic, 2020-12-03:
Oh, that's very nice, I must have missed it in the documentation!

Thanks for your help!

fuegovic, 2020-12-03:
A quick note, when I tried with you code I had an error compiling from "Button::FALLING".  
If I use "Button::Falling" instead everything seems to work as intended.

Thanks again!

closed 2020-12-03 by fuegovic

tttapa, 2020-12-04:
Ah yes, my bad, I should have double checked, glad to hear it works now!