tttapa/Control-Surface #350: Send two different ProgramChange by pressing one button in turn.

santta, 2020-12-03:
Hi, Pieter.

I have one button and one LED.
I need to send two different ProgramChange by pressing one button in turn.

(The first press of the button sends PC-1, the second press of the button sends PC-2, the third press again sends PC-1, and loops)

The LED should light up on PC-2 and turn off when the button is pressed again (and send PC-1)

How can this be done?

fuegovic, 2020-12-04:
I would probably do something like this but there must be a better way With my method I end up doing the debouncing myself even though the library could handle it.

Most of the code is for debouncing, the custom midi message part is quite simple. It might not be the most efficient way but I tested it and it works. 

Good luck with your project!

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

int mode_led_pin = 9; // led on pin 9
// debounced button variables
const int btn_pin = 10;    // the number of the pushbutton pin
int btn_state;             // the current reading from the input pin
int last_btn_state = LOW;   // the previous reading from the input pin
unsigned long lastDebounceTime = 0;  // the last time the output pin was toggled
unsigned long debounceDelay = 50;    // the debounce time; increase if the output flickers
bool mode = false;

void setup() {
  Control_Surface.begin();
  pinMode(10, INPUT_PULLUP); //btn
  pinMode(mode_led_pin, OUTPUT); //mode LED
}

void loop() {
  Control_Surface.loop();

  int btn_reading = digitalRead(btn_pin);
  if (btn_reading != last_btn_state) {
    // reset the debouncing timer
    lastDebounceTime = millis();
  }
  if ((millis() - lastDebounceTime) > debounceDelay) {
    // whatever the reading is at, it's been there for longer than the debounce
    // delay, so take it as the actual current state:

    // if the button state has changed:
    if (btn_reading != btn_state) {
      btn_state = btn_reading;

      // only toggle the msg if the new button state is HIGH 
      if (btn_state == HIGH) {
        mode = !mode;
        if (mode == true){
        Control_Surface.sendPC(1);
        analogWrite(mode_led_pin, 0);    
        }
        if (mode == false){
        Control_Surface.sendPC(2);
        analogWrite(mode_led_pin, 255); // my led is on a pwm pin so I can choose the brightness (value between 0 and 255)                   
        }
      }
    }
  }
  // save the reading. Next time through the loop, it'll be the lastButtonState:
  last_btn_state = btn_reading; 
}
```

fuegovic, 2020-12-04:
Check this exemple to have the library handle the button:

[Send-MIDI-Notes_8ino-example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d8d/Send-MIDI-Notes_8ino-example.html)

santta, 2020-12-04:
Hi, thanks for responding.
How add a few of these buttons and LEDs?

fuegovic, 2020-12-04:
I'm away from my PC right now but I'll try to send you another example with 2 buttons and 2 LEDs based on the "Send-MIDI-Notes_8ino-example" when I have time later. You should be able to scale it up quite easily if needed.

fuegovic, 2020-12-04:
This is working. 
If I press button1 the midi message alternate between PC1 and PC2 and the led1 lights up on PC2 and turn off on PC1. 
If I press button2 the midi message alternate between PC3 and PC4 and the led2 lights up on PC4 and turn off on PC3. 

I am a beginner myself so there might be ways to make the code shorter with arrays but this will work and I tried to make everything clear so  you should be able to tell what is what in the code and add/remove/modify it for your needs.

If you use more than a few leds and buttons you should try to learn how to use arrays to shorten your code but it is optional.

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

Button pushbutton1 = {14};                        //button 1 on pin 14
Button pushbutton2 = {15};                        //button 2 on pin 15

int led1pin = 8;                                  //led 1 on pin 8 
int led2pin = 9;                                  //led 2 on pin 9

bool btn1_mode = false;                           //variable that alternate after each push of button 1
bool btn2_mode = false;                           //variable that alternate after each push of button 2

void setup() {
  Control_Surface.begin();
  pushbutton1.begin();                            // enables internal pull-up for button 1
  pushbutton2.begin();                            // enables internal pull-up for button 2
  pinMode(led1pin, OUTPUT);                       //set led 1 pin as output
  pinMode(led2pin, OUTPUT);                       //set led 2 pin as output
}

void loop() {
  Control_Surface.loop();
  pushbutton1.update();                           // read button 1 state
  pushbutton2.update();                           // read button 2 state
  
  if (pushbutton1.getState() == Button::Falling){ //when button 1 is pressed
    btn1_mode = !btn1_mode;                       //change the state of the variable after each press 
       
    if (btn1_mode == true){                       //if variable is true 
      Control_Surface.sendPC(1);                  //send Program Change 2
      analogWrite(led1pin, 0);                    //led1 turn OFF  
    }
    if (btn1_mode == false){                      //if variable is false 
      Control_Surface.sendPC(2);                  //send Program Change 2
      analogWrite(led1pin, 255);                  //led2 turn ON                  
    }
  }
  
  if (pushbutton2.getState() == Button::Falling){  //when button 2 is pressed
    btn2_mode = !btn2_mode;                        //change the state of the variable after each press 
       
    if (btn2_mode == true){                        //if variable is true 
      Control_Surface.sendPC(3);                   //send Program Change 3
      analogWrite(led2pin, 0);                     //led2 turn OFF  
    }
    if (btn2_mode == false){                       //if variable is false 
      Control_Surface.sendPC(4);                   //send Program Change 4
      analogWrite(led2pin, 255);                   //led1 turn ON 
    }
  }  
}
```

santta, 2020-12-04:
I assembled the circuit on a breadboard and everything works fine, thanks for the help

santta, 2020-12-04:
Is it possible to  turn off LED # 1 when pressing button # 2, and so on, in a loop?

fuegovic, 2020-12-04:
Yes, you just have to move the led variable.

This the part that turn the led 1 off
```cpp
      analogWrite(led1pin, 0);
```

if you replace 
```cpp
      analogWrite(led2pin, 0);
```

with
```cpp
      analogWrite(led1pin, 0);
```

it's the led1 that will turn off instead.

if you add it like that:
```cpp
if (btn2_mode == true){            //if variable is true 
      Control_Surface.sendPC(3);   //send Program Change 3
      analogWrite(led2pin, 0);     //led2 turn OFF
      analogWrite(led1pin, 0);     //led1 turn OFF
    }
```
both leds will turn off.

If you want them to turn back on you will need to make similar change in the part with the other PC message but with 255 (for full brightness) instead of 0 (led off)

Try to move stuff around to better understand the code, you won't break anything if you make a mistake ;)

closed 2020-12-06 by santta

santta, 2020-12-08:
I was able to set up the code this way:

    #include <Control_Surface.h>

    USBMIDI_Interface midi;

     Button pushbutton1 = {2};                        //button 1 on pin 14
     Button pushbutton2 = {3};                        //button 2 on pin 15

      int led1pin = 10;                                  //led 1 on pin 8 
       int led2pin = 11;                                  //led 2 on pin 9

     bool btn1_mode = false;                           //variable that alternate after each push of button 1
     bool btn2_mode = false;                           //variable that alternate after each push of button 2

    void setup() {
    Control_Surface.begin();
    pushbutton1.begin();                            //enables internal pull-up for button 1
    pushbutton2.begin();                            //enables internal pull-up for button 2
    pinMode(led1pin, OUTPUT);                       //set led 1 pin as output
    pinMode(led2pin, OUTPUT);                       //set led 2 pin as output
    }

     void loop() {
     Control_Surface.loop();
     pushbutton1.update();                           // read button 1 state
     pushbutton2.update();                           // read button 2 state
  
     if (pushbutton1.getState() == Button::Falling){ //when button 1 is pressed
      btn1_mode = !btn1_mode;                       //change the state of the variable after each press 
       
      if (btn1_mode == true){                       //if variable is true 
       Control_Surface.sendPC(1);                  //send Program Change 2
       analogWrite(led2pin, 0);                    //led2 turn OFF
       analogWrite(led1pin, 255);                  //led1 turn ON                
       }
      if (btn1_mode == false){                     //if variable is false 
       Control_Surface.sendPC(2);                 //send Program Change 2
       analogWrite(led2pin, 0);                   //led2 turn OFF
       analogWrite(led1pin, 0);                   //led1 turn OFF                 
       }
       }
  
       if (pushbutton2.getState() == Button::Falling){  //when button 2 is pressed
       btn2_mode = !btn2_mode;                        //change the state of the variable after each press 
       
      if (btn2_mode == true){                          //if variable is true 
      Control_Surface.sendPC(3);                 //send Program Change 3
      analogWrite(led2pin, 255);                 //led2 turn ON
      analogWrite(led1pin, 0);                   //led1 turn OFF
       }
    
    if (btn2_mode == false){                       //if variable is false 
      Control_Surface.sendPC(4);                   //send Program Change 4
      analogWrite(led2pin, 0);                     //led2 turn OFF
      analogWrite(led1pin, 0);                     //led1 turn OFF                 
      }
     }  
     }

fuegovic, 2020-12-08:
No problem, glad it worked out for you!