tttapa/Control-Surface #352: ssd1306 don t boot every time

benwadub, 2020-12-06:
hi, do you know why my ssd1306 don t start every time I boot the teensy 4.0? I often have to reboot the teensy for my ssd1306 spirit display start

```cpp

// ----------------------------- Display setup ------------------------------ //
// ========================================================================== //

/*
   Instantiate and initialize the SSD1306 OLED display
*/

constexpr uint8_t SCREEN_WIDTH = 128;
constexpr uint8_t SCREEN_HEIGHT = 64;

constexpr int8_t OLED_DC = 22;    // Data/Command pin of the display
constexpr int8_t OLED_reset = -1; // Use the external RC circuit for reset
constexpr int8_t OLED_CS = 10;   //wired on 13

constexpr uint32_t SPI_Frequency = SPI_MAX_SPEED;

// Instantiate the displays
Adafruit_SSD1306 ssd1306Display = {
  SCREEN_WIDTH, SCREEN_HEIGHT, &SPI,          OLED_DC,
  OLED_reset,   OLED_CS,       SPI_Frequency,
};

// --------------------------- Display interface ---------------------------- //
// ========================================================================== //

// Implement the display interface, specifically, the begin and drawBackground
// methods.
class MySSD1306_DisplayInterface : public SSD1306_DisplayInterface {
 public:
  MySSD1306_DisplayInterface(Adafruit_SSD1306 &display)
    : SSD1306_DisplayInterface(display) {}

  void begin() override {
    delay(200);
    // Initialize the Adafruit_SSD1306 display
    if (!disp.begin())
      FATAL_ERROR(F("SSD1306 allocation failed."), 0x1306);

    // If you override the begin method, remember to call the super class method
    SSD1306_DisplayInterface::begin();
  }

  void drawBackground() override { disp.drawLine(1, 8, 126, 8, WHITE); }

} display = ssd1306Display;
```

tttapa, 2020-12-08:
How are you resetting the displays? Try using an output pin to reset the display, as shown in the Adafruit_SSD1306 examples.  
Is this a problem when using just the Adafruit_SSD1306 library directly, or only when using Control Surface?

benwadub, 2020-12-08:
I reset with a 10k resistor between reset  pin and ground as explained in an exemple in your library, I didn’t try the ssd1306 with another library

tttapa, 2020-12-08:
> I reset with a 10k resistor between reset pin and ground

That won't work, you need a capacitor between the reset pin of the display and ground, and a resistor from reset to 3.3V.

> I didn’t try the ssd1306 with another library

Control Surface just uses the Adafruit_SSD1306 library. Try your setup with their examples, if it has the same issues, then this is probably an issue with your hardware or wiring, not an issue with Control Surface.

henkmeid, 2021-01-30:
@benwadub my experience is that it works the best using a IO pin for reset, if you can spare one, it worked out the most reliable for me.

I use multiple screen and i use a separate IO pin for each screen. It might be eating up a lot of IO pins, but every time i start up my project al screen boot up without any glitches.

benwadub, 2021-01-30:
The problem is that the screen was added to a very big project where lots of pins were already used!
A friend built the same with same pcbs I built and he doesn’t have problem, I think I ll have to check m’y wiring

henkmeid, 2021-01-30:
Also what I found is that when using a lot of wires (breadboard and stuff), it can cause a lot of interferences, it all becomes more stable when everything is routed through a pcb. But I also understand you want to make sure everything works before getting pcb's made 🤔