tttapa/Control-Surface #353: CC parameter selection from menu system.

Gandrasg, 2020-12-07:
Hi tttapa!
I want to make a programmable controller based on your library. I want to be able to set the MIDI CC parameter I want to control on the controller using a menu system. I want to provide access to practically all midi cc parameters, not only would I like to put fixed cc parameters in the banks, but I would also like to set the contents of the banks on the controller. Do you think this is possible with your library? Can the Bankable :: ManyAddresses class be suitable for this purpose? Is it possible to populate this class from an array that contains all CC parameters (0-127)?
Thanks in advance. Sincerely. Andrew

tttapa, 2020-12-08:
While this is possible by using custom banks, it's probably easiest to do this manually.

You can use the following examples to get started:

- https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d3b/Custom-MIDI-Output-Element_8ino-example.html
- https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d6b/Custom-MIDI-Output-Element-Bankable_8ino-example.html
- https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d8d/Send-MIDI-Notes_8ino-example.html

Gandrasg, 2020-12-10:
Thank you for the advices.
I'm testing. What do you mean by manual? Is uploading from an array a bad idea? 

My design is an IOS MIDI controller, 4 bank select buttons, 4 rotary encoders (or potentiometer), and 8 clip launcher (note) buttons. So I will have a total of 16 RE (or pot) and 32 buttons. I would like to implement the display (menu system) on an ILI9341 (or more SD1306) display. I want to set the parameterization (cc and note) of these design controls and I also want to add a visual feedback in barchart form. I've already done simpler tests. 
I am now testing the Bankable: CCAbsolutEncoser branch. Do you have any more advice on what has been described? 
I would be grateful.

tttapa, 2020-12-21:
By manual, I mean that you compute the MIDI addresses yourself, based on your configuration, rather than using the Control Surface banking system.

The `new-input` branch is newer than the `share-encoder` branch, I'd recommend using that one instead, I don't recall exactly, but there might be some issues with `share-encoder` that were fixed in `new-input`.

Gandrasg, 2020-12-22:
thanks for the advice