tttapa/Control-Surface #355: RotaryEncoder Example not compiling

MoetMoe, 2020-12-09:
Hi, 

first of all thanks a lot for the nice work!

I'm trying to get my rotary encoder to function, unfortunately without success so far.
When I try to compile the RotaryEncoder example, I'm getting the following error:

''''

Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Build options changed, rebuilding all
In file included from C:\Users\Mo\Documents\Arduino\libraries\Control-Surface-master\examples\1. MIDI Output\3. Rotary Encoders\RotaryEncoder\RotaryEncoder\RotaryEncoder.ino:37:0:

C:\Users\Mo\Documents\Arduino\libraries\Encoder/Encoder.h:41:37: fatal error: utility/direct_pin_read.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

''''

I tried different IDE versions (1.8.13, 1.8.12, 1.8.9) and different boards (Mega, Uno, Nano etc.). Every other example sketch I tried did work well with my Mega.

Thanks a lot for the help,
Moritz

tttapa, 2020-12-09:
It sounds like you haven't installed the PJRC Encoder library correctly. Please see https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/da8/md_pages_Installation.html.

MoetMoe, 2020-12-10:
That was it! Now it works! Thanks!

closed 2020-12-10 by MoetMoe