tttapa/Control-Surface #358: Midi a serial y ble independientes

alomdj00, 2020-12-10:
Hola Peter.

 Bien, no se si sea posible direccionar los mensajes midi a serial y ble independientes, es decir, digamis que los faders solo funcionen via serial o midi out, y los encoders solo via ble, y al mismo tiempo, asi tener 2 controladores por separado,  como si fuese un tensy, no se si me explique correctamente, disculpa si suena algo enredado esto. 

De esta forma saldria mas economico que una tensy, al menos en mi ciudad con lo de una teensy compro seis esp32. O me sobra para los potenciometros, encoders y una oled.

Muchas gracias por tu trabajo.

tttapa, 2020-12-16:
You could try something like this:

```cpp
#include <Control_Surface.h>

// Pipe that streams MIDI messages from a MIDI source to a MIDI sink,
// filtering the messages based on their cable number.
class CableFilterMIDI_Pipe : public MIDI_Pipe {
  public:
    CableFilterMIDI_Pipe(Cable from, Cable to = CABLE_1) : from(from), to(to) {}

    void mapForwardMIDI(ChannelMessage msg) override { 
      if (msg.getCable() == from) {
        msg.setCable(to);
        sourceMIDItoSink(msg); 
      }
    }

    void mapForwardMIDI(SysExMessage msg) override { 
      if (msg.getCable() == from) {
        msg.setCable(to);
        sourceMIDItoSink(msg); 
      }
    }

    void mapForwardMIDI(RealTimeMessage msg) override { 
      if (msg.getCable() == from) {
        msg.setCable(to);
        sourceMIDItoSink(msg); 
      }
    }

  private:
    Cable from, to;
};

HardwareSerialMIDI_Interface ser { Serial };
BluetoothMIDI_Interface ble;

// All data on cable 1 will be sent to cable 1 of the Serial MIDI interface
CableFilterMIDI_Pipe cs2ser { CABLE_1, CABLE_1 };
// All data received on cable 1 of the Serial MIDI interface will be sent to cable 1 of Control Surface
CableFilterMIDI_Pipe ser2cs { CABLE_1, CABLE_1 };

// All data on cable 2 will be sent to cable 1 of the BLE MIDI interface
CableFilterMIDI_Pipe cs2ble { CABLE_2, CABLE_1 };
// All data received on cable 1 of the BLE MIDI interface will be sent to cable 2 of Control Surface
CableFilterMIDI_Pipe ble2cs { CABLE_1, CABLE_2 };

CCButton btn_ser { 2, {0x10, CABLE_1} }; // This button sends to cable 1 (i.e. Serial MIDI)
CCButton btn_ble { 3, {0x10, CABLE_2} }; // This button sends to cable 2 (i.e. BLE MIDI)

void setup() {
  // Do the MIDI routing first
  Control_Surface >> cs2ble >> ble; // Control Surface output to BLE MIDI
  Control_Surface << ble2cs << ble; // BLE MIDI input to Control Surface
  
  Control_Surface >> cs2ser >> ser;
  Control_Surface << ser2cs << ser;

  // Then initialize Control Surface
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

alomdj00, 2020-12-16:
Es grandioso,  creo que esto me evita comprar una teensy y aprovechar lo que ya tengo.

Tambien me ha surgido una duda. El esp32 tiene tres seriales, el que viene conectado al ch340 o cp2102, y otros dos. Hasta este punto todo perfecto, ahora digamos que voy a usar el serial2, que seria el tercer serial del esp32, lo voy a conectar al serial1 del leonardo para aprovechar su usb nativo. Se que tengo que usar optoacopladores para evitar problemas, pero solo seria eso, es decir no necesito del ch340 y parecidos, solo el circuito de los optoacopladores recomendado por la pagina de midi

tttapa, 2020-12-17:
Optocouplers are used in MIDI equipment to galvanically isolate different devices and prevent ground loops. You don't need them to connect TTL UARTs together. You will need a level shifter, though, because the Arduino Leonardo is a 5V device and the ESP32 is a 3.3V device.

alomdj00, 2020-12-17:
Vale, empezare a hacer pruebas.
Muchisimas gracias por despejarme  esta duda. Un saludo

alomdj00, 2020-12-25:
Hola Peter, me preguntaba si esto tambien funciona para Wifi,  cable 1 a wifi,  cable 2 a serial....

En el ejemplo de wifi esta es la parte que define la  interface segun entendi:

// ----------------------------- MIDI Interface ----------------------------- //

// First create the AppleMIDI instance
APPLEMIDI_CREATE_INSTANCE(WiFiUDP, MIDI, "ESP32", 5004);
//                           │       │      │       └──── Local port number
//                           │       │      └──────────── Name
//                           │       └─────────────────── MIDI instance name
//                           └─────────────────────────── Network socket class

// Then wrap it in a Control Surface-compatible MIDI interface
FortySevenEffectsMIDI_Interface<decltype(MIDI) &> AppleMIDI_interface = MIDI;

Sin embargo no se cual es la interface a usar es el filtrado, es decir segun lo anterior, cual es la palabra o definicion que reemplazaria a ble, en tu ejemplo que posteaste aqui

No se si me explique correctamente, solo posteo esa parte del codigo, ya que considero que lo demas se mantiene o construye normalmente, mi duda radica en el nombre de la interfaz midi wifi a usar en el filtrado.

Por tu respuesa u orientacion estare infinitamente agradecido. Un saludo.

tttapa, 2020-12-25:
You have to use the Control Surface interface, i.e. `AppleMIDI_interface`.