tttapa/Control-Surface #359: ESP32 Display on SSD1306 (SPI 7wire)

Gandrasg, 2020-12-12:
Hi ttapa!

Testing this code: _Control-Surface/examples/7.Displays/MCU-OLED-SSD1306/MCU-OLED-SSD1306.ino_
on my ESP32.
I can see that the code has not been tested on ESP32, only on Teensy. Do you know that you have successfully applied it to ESP32?
Give me this error:
```
_"exit status 1
'class SPIClass' has no member named 'setMOSI' "_
```

tttapa, 2020-12-15:
You can just leave out the line with the `setMOSI` statement. It's a Teensy-specific function that selects an alternative SPI pin, you probably don't need to do that (only if there are conflicts with other pins you want to use).

Gandrasg, 2020-12-17:
I understand. Thanks, I'm going to do it.