tttapa/Control-Surface #360: Buttons and potentiometers on one multiplexer

LukaszChrzanowski, 2020-12-13:
Hello

is there possibility to use on one multiplexer buttons and rotary? 

In eg. I have this code:

```
CD74HC4067 mux = {
  A9,       // Analog input pin
  {3,2,0,1}
};

//----------- KNOBS -----------
CCPotentiometer volumePotentiometers[] = {
  { mux.pin(0), { 0x07, CHANNEL_1 } },
  { mux.pin(1), { 0x07, CHANNEL_2 } },
  { mux.pin(2), { 0x0E, CHANNEL_2 } },
  { mux.pin(3), { 0x0E, CHANNEL_1 } },
  { mux.pin(4), { 0x0A, CHANNEL_1 } },
  { mux.pin(5), { 0x0A, CHANNEL_2 } },
  { mux.pin(6), { 0x0F, CHANNEL_1 } },
  { mux.pin(7), { 0x0A, CHANNEL_2 } },
  { mux.pin(8), { 0x0E, CHANNEL_2 } },
  { mux.pin(9), { 0x0F, CHANNEL_2 } },
  { mux.pin(10), { 0x0A, CHANNEL_3 } },
  { mux.pin(11), { 0x0A, CHANNEL_4 } },
  { mux.pin(12), { 0x0E, CHANNEL_3 } },
  { mux.pin(13), { 0x0E, CHANNEL_4 } },
  { mux.pin(14), { 0x0F, CHANNEL_4 } },
  { mux.pin(15), { 0x0F, CHANNEL_3 } },
};

```
but I would like have on pins 8-15 buttons.

Is there a way? :)

LukaszChrzanowski, 2020-12-13:
ok, I think I've got solution (I think):

`CD74HC4067 mux = {
  A9,       // Analog input pin
  {3,2,0,1}
};

//----------- KNOBS -----------
CCPotentiometer volumePotentiometers[] = {
  { mux.pin(0), { 0x07, CHANNEL_1 } },
  { mux.pin(1), { 0x07, CHANNEL_2 } },
  { mux.pin(2), { 0x0E, CHANNEL_2 } },
  { mux.pin(3), { 0x0E, CHANNEL_1 } },
  { mux.pin(4), { 0x0A, CHANNEL_1 } },
  { mux.pin(5), { 0x0A, CHANNEL_2 } },
  { mux.pin(6), { 0x0F, CHANNEL_1 } },
  { mux.pin(7), { 0x0A, CHANNEL_2 } },
  { mux.pin(8), { 0x0E, CHANNEL_2 } },
  { mux.pin(9), { 0x0F, CHANNEL_2 } },
};

CCButton buttonMySuperMatrix[] = {
  { mux.pin(10), { 0x0A, CHANNEL_3 } },
  { mux.pin(11), { 0x0A, CHANNEL_4 } },
  { mux.pin(12), { 0x0E, CHANNEL_3 } },
  { mux.pin(13), { 0x0E, CHANNEL_4 } },
  { mux.pin(14), { 0x0F, CHANNEL_4 } },
  { mux.pin(15), { 0x0F, CHANNEL_3 } },
};`

but I remember that there was some problem with this approach (its somewhere hardcoded that CCPotentiometer or CCButton have to have 8 or 16 elements...

tttapa, 2020-12-15:
> but I remember that there was some problem with this approach (its somewhere hardcoded that CCPotentiometer or CCButton have to have 8 or 16 elements...

That's not the case, you can have any number of CCPotentiometer and CCButton objects. The code in your previous reply looks fine to me.

You cannot connect your rotary encoders to multiplexer pins, though.

LukaszChrzanowski, 2020-12-15:
encoders no, but potentiometers yes 👍 

generaly - tried, work as intended :)