tttapa/Control-Surface #364: Enable disable leds

alomdj00, 2020-12-28:
Hola Peter.

Me preguntaba si es posible habilitar y deshabilitar leds, con potenciometros y botones funciona, mientras que con leds si bien compila y sube a la placa no funciona, con el ejemplo que proporcionas de enable, reemplace el potenciometro por un led de MCU::SOLO_1, sin embargo no funciono. Lo que propongo se puede hacer al estilo de un banco o algo silimar

Muchas gracias por tu respuesta. Un saludo

tttapa, 2020-12-28:
You'll have to provide much more information. Please post your code and how you're testing it. The LED classes should work, and there are bankable versions as well, so it's not really clear to me what you're asking?

alomdj00, 2020-12-28:
Este es el codigo,   lo he escrito asi

`

#include <Control_Surface.h>

USBMIDI_Interface midi;

Button enableButton = 2; // Momentary push button between pin 2 and ground

using namespace MCU;

NoteLED led = {
  LED_BUILTIN,             // Pin of built-in LED
  {MCU::SOLO_1}, // SOLO FADER UNO MACKIE
};

void setup() {
  enableButton.begin();
  Control_Surface.begin();
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);
}

void loop() {
  if (enableButton.getState() == Button::Falling) { // al presionar el boton el led debe apagarse, al soltar debe volver a encender si esta activo
    static bool state = true;
    state = !state;
    digitalWrite(LED_BUILTIN, state);
    state ? led.enable() : led.disable();
  }
  Control_Surface.loop();
}

`

El led tendria  queapagarse  presionar el led, mi idea es utilizarlo para seleccionar bancos por ejemplo, a fin de poder reutilizar leds, y que no quede tan grande.

Muchas gracias

tttapa, 2020-12-28:
`NoteLED::enable()` and `NoteLED::disable()` only determine whether the LED responds to incoming MIDI messages or not, they don't change the state of the LED itself.

If you want to do more advanced things with your LEDs, I'd recommend using the [`NoteValue`](https://tttapa.github.io/Control-Surface-doc/Doxygen/de/ddd/classNoteValue.html) class instead.  
If you aren't already, you might want to use the `new-input` branch, which allows you to check when the value of a `NoteValue` changed. See [`new-input: NoteValue`](https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/df/d8b/group__MIDIInputElements.html#ga8a39b5a02a5f65c3373150ab55740f4c)

For example:

```cpp
NoteValue solo1 = {MCU::SOLO_1}; // SOLO FADER UNO MACKIE
const pin_t solo1_led = LED_BUILTIN;

void setup() {
  pinMode(solo1_led, OUTPUT);
  Control_Surface.begin();
}
...
void loop() {
  Control_Surfac.loop();
  if (solo1.getDirty()) { // Check the dirty flag (the dirty flag is set to 1 when the MIDI value changes)
    // The MIDI value has changed, update the state of the LED
    bool ledState = solo1.getValue() > 0;
    // Add your own logic here, e.g. change the LED state based on a button press
    digitalWrite(solo1_led, ledState ? HIGH : LOW);
    solo1.clearDirty(); // Reset the dirty flag to 0, ready to detect the next MIDI value change
  }
}
```
The `NoteValue` class just listens for incoming MIDI messages, you update the state of the LED yourself in the main loop.