tttapa/Control-Surface #371: Two buttons - four CC

Wallirec, 2021-01-01:
Hello, I'm a beginner and just learning arduino. Help me write a sketch using this library for a controller on two buttons that sends different CC messages with a short and long press (CC 20 - CC 21, CC 22 - CC 23).

tttapa, 2021-01-01:
This is all I have time for at the moment. Adding a second button should be easy enough, for more buttons, you probably want to use an array and some for loops, or wrap it in a class. 

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

// Ignore the name “increment” button, it's simply a button
// class that can detect long and short presses.
IncrementButton btn { 3 }; // pin number of button

void setup() {
  Control_Surface.begin();
  btn.begin();
}

void loop() {
  Control_Surface.loop();
  switch (btn.update()) {
    case IncrementButton::ReleasedShort: // If button is released after short press
      Control_Surface.sendCC(20, 127); break;
    case IncrementButton::IncrementLong: // If button is still pressed after some time
      Control_Surface.sendCC(21, 127); break;
  }
}
```

Wallirec, 2021-01-01:
Thanks, it works!

Wallirec, 2021-05-24:
Hello! I have successfully used the controller all this time, but now I need to send a CC with a value of 0 on the next press - is it possible?

closed 2021-05-26 by Wallirec