tttapa/Control-Surface #372: Esp8266 wifi no compila

alomdj00, 2021-01-03:
Hola peter, he estado tratando de compilar esp8266 para wifi y me ha resultado imposible, estoy utilizando la ultima version de applemidi desde el repositorio de github, asi como tu ultima version de control surface, probe tanto como con la rama new imput y la rama maestra y me salta el mismo error

```
/**
 * This example demonstrates how to use the AppleMIDI library to use Control
 * Surface over the network.
 * 
 * This example requires the 
 * [AppleMIDI](https://github.com/lathoub/Arduino-AppleMIDI-Library) and 
 * [MIDI](https://github.com/FortySevenEffects/arduino_midi_library) libraries.
 * 
 * @boards  ESP32
 * 
 * Connections
 * -----------
 * 
 * The on-board LED will be used, as well as the push button on GPIO0. These 
 * should be present on most ESP32 boards, if this is not the case for your 
 * board, connect an LED (+ series resistor) and a push button to the relevant
 * pins (in the "MIDI Elements" section below). For more details, see the 
 * @ref NoteButton.ino and @ref 1.Note-LED.ino examples.
 * 
 * WiFi Credentials
 * ----------------
 * 
 * Open the tab `WiFi-Credentials.example.h`, enter your WiFi credentials, and
 * rename the file to `WiFi-Credentials.h`.
 * 
 * Behavior
 * --------
 * 
 * Upload the code to the ESP32, and open the Serial monitor. You should see
 * output like this
 * 
 * ~~~
 * Connecting to Your WiFi Network ...
 * Connected!
 * IP address: 192.168.1.35
 * mDNS responder started (ESP32.local)
 * ~~~
 * 
 * Next, connect to the device using your DAW or other MIDI software. If the 
 * software supports mDNS (Apple Bonjour), you can use `ESP32.local`, 
 * otherwise, you'll have to use the IP address.
 * 
 * When the connection is successful, you'll see the following message in the
 * Serial monitor:
 * 
 * ~~~
 * Connected to session Your Session Name
 * ~~~
 * 
 * When the button is pushed, a MIDI note on message for note C4 (middle C) is
 * sent.  
 * When the ESP32 receives a MIDI note message for that note, it turn on/off the
 * LED accordingly.
 * 
 * RTP MIDI Bridge (Linux)
 * -----------------------
 * 
 * If you're on Linux, you can use the rtpmidi-bridge application in the example
 * folder. You'll need to install Node.js and NPM.
 * 
 * First, install the necessary dependencies and build tools:
 * 
 * ~~~sh
 * sudo apt install build-essential libasound2-dev libavahi-compat-libdnssd-dev
 * ~~~
 * 
 * Then install the dependencies using NPM:
 * 
 * ~~~sh
 * npm install
 * ~~~
 * You might get a compilation error for the `avahi_pub` module. This is not an
 * issue, it's an optional dependency of the `rtpmidi` module.
 * 
 * Finally, run the application:
 * 
 * ~~~sh
 * node rtpmidi-bridge.js
 * ~~~
 * 
 * The application will initiate a RTP MIDI connection with the ESP32, create
 * virtual MIDI ports, and bridge the MIDI traffic between the RTP MIDI 
 * connection and the virtual MIDI ports.  
 * You can then connect your DAW or other MIDI application to the virtual MIDI 
 * ports.
 * 
 * When the ESP32 is connected, you should see the following in the serial 
 * monitor and the rtpmidi-bridge output respectively:
 * 
 * ~~~
 * Connected to session Node RTPMidi
 * ~~~
 * 
 * ~~~
 * 2020-05-06T15:50:42.956Z info:  Invitation accepted by ESP32
 * 2020-05-06T15:50:42.962Z info:  Data channel to ESP32 established
 * ~~~
 * 
 * Mapping
 * -------
 * 
 * Connect the virtual MIDI ports or the AppleMIDI connection to a device or 
 * application that can send and receive MIDI notes.
 * 
 * Written by PieterP, 2020-05-06  
 * https://github.com/tttapa/Control-Surface
 */

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <WiFiUdp.h>

#include <AppleMIDI.h>
USING_NAMESPACE_APPLEMIDI

#include <Control_Surface.h>
#include <MIDI_Interfaces/Wrappers/FortySevenEffects.hpp>

#include <ESP8266mDNS.h>
#include <WiFi.h>
//#include <WiFiUdp.h>

//#include <WiFiUDP.h>

#include "WiFi-Credentials.example.h" // See instructions above

// ----------------------------- MIDI Interface ----------------------------- //

// First create the AppleMIDI instance
APPLEMIDI_CREATE_INSTANCE(WiFiUDP, MIDI, "ESP8266", 5004);
//                           │       │      │       └──── Local port number
//                           │       │      └──────────── Name
//                           │       └─────────────────── MIDI instance name
//                           └─────────────────────────── Network socket class

// Then wrap it in a Control Surface-compatible MIDI interface
FortySevenEffectsMIDI_Interface<decltype(MIDI) &> AppleMIDI_interface = MIDI;

// ------------------------------ MIDI Elements ----------------------------- //

// Add some MIDI elements for testing
using namespace MIDI_Notes;
NoteButton button = {
    9, note(C, 4),  // GPIO0 has a push button connected on most boards
};

// --------------------------- AppleMIDI callbacks -------------------------- //

void onAppleMidiConnected(const ssrc_t &ssrc, const char *name) {
  Serial << F("Connected to session ") << name << endl;
}
void onAppleMidiDisconnected(const ssrc_t &ssrc) {
  Serial << F("Disconnected") << endl;
}
void onAppleMidiError(const ssrc_t &ssrc, int32_t err) {
  Serial << F("Exception ") << err << F(" from ssrc 0x") << hex << ssrc << dec
         << endl;
}

// ---------------------------------- Setup --------------------------------- //

void setup() {
  Serial.begin(115200);

  // Connect to the WiFi network
  Serial << endl << F("Connecting to ") << ssid << ' ';
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
    Serial.print("."), delay(250);
  Serial << endl
         << F("Connected!") << endl
         << F("IP address: ") << WiFi.localIP() << endl;

  // Set up mDNS responder:
  if (!MDNS.begin(AppleMIDI.getName()))
    FATAL_ERROR(F("Error setting up MDNS responder!"), 0x0032);
  Serial << F("mDNS responder started (") << AppleMIDI.getName() << ".local)" 
         << endl;
  MDNS.addService("apple-midi", "udp", AppleMIDI.getPort());

  // Set up some AppleMIDI callback handles
  AppleMIDI.setHandleConnected(onAppleMidiConnected);
  AppleMIDI.setHandleDisconnected(onAppleMidiDisconnected);
  AppleMIDI.setHandleError(onAppleMidiError);

  // Initialize Control Surface (also calls MIDI.begin())
  Control_Surface.begin();
}

// ---------------------------------- Loop ---------------------------------- //

void loop() {
  // Update all MIDI elements and handle incoming MIDI
  Control_Surface.loop();
}

```

Aqui el error

C:\Users\Alomi\AppData\Local\Temp\arduino_modified_sketch_722356\AppleMIDI.ino: In function 'void setup()':

AppleMIDI:186:13: error: 'class appleMidi::AppleMIDISession<WiFiUDP>' has no member named 'setHandleError'

   AppleMIDI.setHandleError(onAppleMidiError);

             ^

exit status 1
'class appleMidi::AppleMIDISession<WiFiUDP>' has no member named 'setHandleError'

Muchas gracias por tu ayuda.

tttapa, 2021-01-03:
Please see the AppleMIDI documentation: Arduino-AppleMIDI-Library/wiki/Migration-from-2.x-to-3.x](https://github.com/lathoub/Arduino-AppleMIDI-Library/wiki/Migration-from-2.*-to-3.*)

I have no time to fix the Control Surface example right now, but you can just remove the problematic line from your setup (or use the 2.x version of the AppleMIDI library).

alomdj00, 2021-01-03:
He descargado la version 2.2 de apple midi y ahora me ha tirado mas errores.

Por lo pronto seguire con el diseño del pcb, espero puedas arreglarlo mas adelante. 

Aqui dejo los errores que tira ahora.

In file included from C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:29:0:

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.h:27:7: error: redefinition of 'class WiFiUDP'

 class WiFiUDP : public UDP {

       ^

In file included from C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\utility/wifi_drv.h:26:0,

                 from C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:26:

C:\Users\Alomi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src/WiFiUdp.h:32:7: error: previous definition of 'class WiFiUDP'

 class WiFiUDP : public UDP, public SList<WiFiUDP> {

       ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In constructor 'WiFiUDP::WiFiUDP()':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:35:22: error: class 'WiFiUDP' does not have any field named '_sock'

 WiFiUDP::WiFiUDP() : _sock(NO_SOCKET_AVAIL) {}

                      ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In member function 'virtual uint8_t WiFiUDP::begin(uint16_t)':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:45:9: error: '_sock' was not declared in this scope

         _sock = sock;

         ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:46:9: error: '_port' was not declared in this scope

         _port = port;

         ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In member function 'virtual int WiFiUDP::available()':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:56:7: error: '_sock' was not declared in this scope

   if (_sock != NO_SOCKET_AVAIL)

       ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In member function 'virtual void WiFiUDP::stop()':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:66:8: error: '_sock' was not declared in this scope

    if (_sock == NO_SOCKET_AVAIL)

        ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:69:26: error: '_sock' was not declared in this scope

    ServerDrv::stopClient(_sock);

                          ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In member function 'virtual int WiFiUDP::beginPacket(IPAddress, uint16_t)':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:88:7: error: '_sock' was not declared in this scope

   if (_sock == NO_SOCKET_AVAIL)

       ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:90:7: error: '_sock' was not declared in this scope

   if (_sock != NO_SOCKET_AVAIL)

       ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In member function 'virtual int WiFiUDP::endPacket()':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:101:32: error: '_sock' was not declared in this scope

  return ServerDrv::sendUdpData(_sock);

                                ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In member function 'virtual size_t WiFiUDP::write(const uint8_t*, size_t)':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:111:27: error: '_sock' was not declared in this scope

  ServerDrv::insertDataBuf(_sock, buffer, size);

                           ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In member function 'virtual int WiFiUDP::read()':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:125:23: error: '_sock' was not declared in this scope

    ServerDrv::getData(_sock, &b);

                       ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In member function 'virtual int WiFiUDP::read(unsigned char*, size_t)':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:137:31: error: '_sock' was not declared in this scope

    if (!ServerDrv::getDataBuf(_sock, buffer, &size))

                               ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In member function 'virtual int WiFiUDP::peek()':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:152:22: error: '_sock' was not declared in this scope

   ServerDrv::getData(_sock, &b, 1);

                      ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In member function 'virtual IPAddress WiFiUDP::remoteIP()':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:166:25: error: '_sock' was not declared in this scope

  WiFiDrv::getRemoteData(_sock, _remoteIp, _remotePort);

                         ^

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp: In member function 'virtual uint16_t WiFiUDP::remotePort()':

C:\Users\Alomi\Desktop\arduino-1.8.12\libraries\WiFi\src\WiFiUdp.cpp:176:25: error: '_sock' was not declared in this scope

  WiFiDrv::getRemoteData(_sock, _remoteIp, _remotePort);

                         ^

exit status 1
Error compilando para la tarjeta NodeMCU 1.0 (ESP-12E Module).

Muchas gracias. Un saludo

tttapa, 2021-01-03:
Your library setup is messed up. The WiFi library is part of the ESP8266 or ESP32 Core, it should not be in your `libraries` folder at all.