tttapa/Control-Surface #377: Shift register para pines de seleccion en mux

alomdj00, 2021-01-20:
Hola, estoy tratando de usar un muxes con shift regiter para los pines de seleccion  de la siguiente manera

sreg esta definido por la conexion spi.

CD74HC4051 mux = { A0, {sreg.pin(0), sreg.pin(1), sreg.pin(2)};

El mux funciona a medias, es decir, me imprime valores de la siguiente manera:
0.    0.    0.    163.    0.     0.     164.    0

Siempre me tira de dos valores al aplicar voltaje en una sola entrada,  probre el mux con pines directos de arduino y este problema no pasa por lo que el mux no esta dañado.

Probe el shift register 74hc595 por aparte manejando leds y funciona perfecto por lo que el problema tampoco es del shift register.

Tamvien probre definiendo el shift register cln los pines datapin, latchpin y clock pin y con los leds funciona, pero con el muy no funciona

No se si estoy haciendo algo mal, o el shift register es que funciona asi y no soporta esta funcion, no se si hayas probado esto o es que por cuestiones de hardware que no se permite esta configuracion

Tambien probre con MSBFIRST  y LSBFIRST y falla.

Si esto no es posible que alternativas hay para manejar los pines de seleccion de los mux

Muchas gracias. Un saludo

tttapa, 2021-01-20:
Please post the full code.

alomdj00, 2021-01-20:
#include <Control_Surface>

SPIShiftRegisterOut<8> sreg = {SS, MSBFIRST};

CD74HC4051 mux = {A0, {sreg.pin (0), sreg.pin (1), sreg.pin (2)};

void setup() {
  mux.begin();
  Serial.begin(115200);
}

void loop() {
for (pin_t pin = 0; pin < mux.getLength(); ++pin) {
    Serial.print(mux.analogRead(pin));
    Serial.print('\t');
  }
  Serial.println();

}

}

tttapa, 2021-01-20:
You have mismatched braces and you never call `sreg.begin()`.

alomdj00, 2021-01-20:
No se ha que te refieres  con frenos, o al menos asi me lo traduce google,en lo unico que veo que me sobro es un parentesis en el void loop, error mio al copiar y pegar, sin embargo voy a probar con el sreg.begin(), y ya os comento que pasa

tttapa, 2021-01-20:
`(...)` are parentheses, `{...}` are braces.