tttapa/Control-Surface #424: Please implement the undefined CC numbers

craigyjp, 2021-03-07:
**Is your feature request related to a problem? Please describe.**

I have built a MIDI controller to control the Arturia range of softsynths such as the CS80v. The first limit I hit was usable CC#'s as Arturia do not support NRPN, RPN, sysex etc for control. Only CC and that is limited to around 118 due to the way they reserve certain CC numbers. So I asked Arturiato implement NRPN again which they took away in later versions. But this is pipe dream and may never happen.

I then started looking at your library as I wanted to make my controller more universal, support NRPN etc for different sysnths and it seems so easy to use I thought I would give it a go. Then I realised very quickly you do not support the Undefined CC numbers. This limiting the range to about 70 available CC numbers.

**Describe the solution you'd like**

Is it possible to simply inplement Undefined_3 for example for all the undefined CC numbers and allow them to be sent?
It seems to me like an easy solution.

**Describe alternatives you've considered**
I have no alternatives except to carry on with my own code because of this limitation.

**Additional context**
Add any other context or code about the feature request here.

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;  // Instantiate a MIDI Interface to use
 
// Instantiate an analog multiplexer
CD74HC4067 mux1 = {
  A2,       // Analog input pin
  {28, 32, 30, 31} // Address pins S0, S1, S2
};
 
// Create an array of potentiometers that send out MIDI Control Change messages 
// when you turn the potentiometers connected to the 8 input pins of the mux.
CCPotentiometer volumePotentiometers[] = {
  { mux1.pin(0), { MIDI_CC::Breath_Controller, CHANNEL_1 } },
  { mux1.pin(1), { MIDI_CC::Undefined_3, CHANNEL_1 } },
  { mux1.pin(2), { MIDI_CC::General_Purpose_Controller_3, CHANNEL_1 } },
  { mux1.pin(3), { MIDI_CC::General_Purpose_Controller_4, CHANNEL_1 } },
  { mux1.pin(4), { MIDI_CC::General_Purpose_Controller_5, CHANNEL_1 } },
  { mux1.pin(5), { MIDI_CC::General_Purpose_Controller_6, CHANNEL_1 } },
  { mux1.pin(6), { MIDI_CC::General_Purpose_Controller_8, CHANNEL_1 } },
  { mux1.pin(7), { MIDI_CC::Expression_Controller, CHANNEL_1 } },
};

void setup() {
  Control_Surface.begin();  // Initialize the Control Surface

}

void loop() {
  Control_Surface.loop();  // Update the Control Surface

}
```

tttapa, 2021-03-07:
You don't have to use the named constants, simply use the actual CC number. For example, replacing `MIDI_CC::Breath_Controller` by `0x02` or `2` would give exactly the same results. The names in the MIDI_CC namespace are not magic, they are just ordinary constants, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/dbe/namespaceMIDI__CC.html.

craigyjp, 2021-03-07:
Well I tried that and it worked pefectly well, thankyou. I will now try and code the full 128 inputs (less of course the reserved CC numbers).