tttapa/Control-Surface #425: Examples cant find file - Encoder.h

dchstudios, 2021-03-07:
A number of examples make use of including <Encoder.h> at the beginning
 - e.g  -> MCU-OLED-SSD1306 and MCU-OLED-SSD1306-MEGA
However, compiling in Arduino errors out 
= no such file or directory Encoder.h

If I comment out the inclusion of Encoder.h, then I get this error: 

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Seeeduino XIAO, Arduino, Off"

C:\Users\nwasp\Documents\Arduino\MCU-OLED-SSD1306\MCU-OLED-SSD1306.ino: In function 'void setup()':

MCU-OLED-SSD1306:245:7: error: 'class SPIClass' has no member named 'setMOSI'

   SPI.setMOSI(7);

       ^~~~~~~

Multiple libraries were found for "Adafruit_SSD1306.h"

 Used: C:\Users\nwasp\Documents\Arduino\libraries\Adafruit_SSD1306

 Not used: C:\Users\nwasp\Documents\Arduino\libraries\Adafruit_SSD1306-2.4.2

Multiple libraries were found for "Adafruit_ZeroDMA.h"

 Used: C:\Users\nwasp\Documents\ArduinoData\packages\Seeeduino\hardware\samd\1.8.1\libraries\Adafruit_ZeroDMA

 Not used: C:\Users\nwasp\Documents\Arduino\libraries\Adafruit_Zero_DMA_Library

exit status 1

'class SPIClass' has no member named 'setMOSI'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

------------------------------------
My code environment: 
[Arduino vers1.8.13 running on Windows 10]
[board: Seeed XIAO]

- Thanks, David

dchstudios, 2021-03-07:
I commented out that MOSI line as per the suggestion I read in another issue thread.. It then compiled and uploaded.. -> (the MCU-OLED-SSD1306) Example.. just no output to the screen. I'm using an OLED SSD1306 that is similiar to the Adafruit, but not Adafruit's product. Also I'm using Cubase 11 as my DAW and outputting MIDI to the XIAO .  I see that the XIAO is receiving MIDI data via it's onboard LED, just nothing goes to the 1306 OLED with the example code.
I do get the OLED to work with my XIAO in general when using the U8g2 Library, so the OLED does work, just not with your example code. My OLED does have the same 4 wire connectors as the Adafruit and same 128x64 monochrome screen.

tttapa, 2021-03-07:
> A number of examples make use of including <Encoder.h> at the beginning ... no such file or directory Encoder.h

Please see the [installation instructions](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/da8/md_pages_Installation.html).

> error: 'class SPIClass' has no member named 'setMOSI'

The example is intended for Teensy boards, which do have a `setMOSI` function.

> My OLED does have the same 4 wire style connectors as the Adafruit and same 128x64 monochrome screen.

Four wires imply I²C, the example uses SPI. Please see the examples that come with the `Adafruit_SSD1306` library to learn how to use I²C displays.

It's simply a matter of using the correct initialization here:
https://github.com/tttapa/Control-Surface/blob/feee931d44d5f20fb7913b171895316e5819549d/examples/7.Displays/MCU-OLED-SSD1306/MCU-OLED-SSD1306.ino#L70-L87
and passing the correct I²C address to the `Adafruit_SSD1306::begin()` function here:
https://github.com/tttapa/Control-Surface/blob/feee931d44d5f20fb7913b171895316e5819549d/examples/7.Displays/MCU-OLED-SSD1306/MCU-OLED-SSD1306.ino#L100-L102