tttapa/Control-Surface #426: Did you follow the [installation instructions](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/da8/md_pages_Installation.html) for the dependencies?

dchstudios, 2021-03-08:
Did you follow the [installation instructions](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/da8/md_pages_Installation.html) for the dependencies?

Also, don't include MIDI.h, it's a different library, and it might interfere.

_Originally posted by @tttapa in https://github.com/tttapa/Control-Surface/issues/83#issuecomment-573849568_

dchstudios, 2021-03-08:
I still can't get the <Encoder.h> file to be seen when used in the #include code. I tried installing the Teensy libraries that it suggested (all), deleted it's version of Midi.h as per above.  I see that the Encoder.h file is buried in the "hardware" folder of the Teensy ->C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Encoder  , does that make it inaccessible if I'm not using a Teensy hardware board specific? e.g. I'm using a Seeed XIAO so I have the board loaded for that.  So then I tried just copying the whole Encoder file from within the hardware library folder of the Teensy, and put it at the higher level with the other regular Arduino libraries -> C:\Program Files (x86)\Arduino\libraries. Still a no go.   <Encoder.h> "cannot be found"

tttapa, 2021-03-08:
You are not installing it correctly. The process is the same for pretty much all libraries.

Go to the [GitHub page for the Encoder library](https://github.com/PaulStoffregen/Encoder), click the green "Code" button and download the repository as a .ZIP file. Then go to the _Sketch > Include Library > Add .ZIP Library_ menu in the IDE
Then navigate to your downloads directory where you just downloaded the library.
Select it, and click Ok.