tttapa/Control-Surface #429: Condition for sending MIDI

Willame156, 2021-03-15:
First of all, thanks for this great library.
I ask for your help in resolving 2 issues:
1st - "button_01" send CC 2, but if more than 1 second send CC 10;
2nd - Do not read "button_01" if button_02 has pressed

// Your code here
#include <Control_Surface.h>    
#include <MIDI_Interfaces/BluetoothMIDI_Interface.hpp> 

BluetoothMIDI_Interface midi;  

CCButtonLatched button_01 = {15 ,  { 2, CHANNEL_1}, };  // if more than 1 second send midi CC 10
CCButton button_02 = {23 ,  { 4, CHANNEL_1}, };  //Do not read "button_01" if button_02 has pressed

                      
void setup() {
   Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}

//I thank you for your attention

ACuppaJoes, 2021-07-31:
I think this tutorial might help you out with your issue
https://www.instructables.com/Arduino-Dual-Function-Button-Long-PressShort-Press/

Willame156, 2021-12-23:
> Acho que este tutorial pode ajudá-lo com seu problema https://www.instructables.com/Arduino-Dual-Function-Button-Long-PressShort-Press/

Obrigado por sua resposta.
Me perdoe a demora em comentar, acabei esquecendo.

tttapa, 2021-12-23:
> I think this tutorial might help you out with your issue
> https://www.instructables.com/Arduino-Dual-Function-Button-Long-PressShort-Press/

Alternatively, you could use https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d83/classAH_1_1IncrementButton.html, it already handles long and short presses for you.

Willame156, 2021-12-24:
> > Acho que este tutorial pode ajudá-lo com seu problema 
> > https://www.instructables.com/Arduino-Dual-Function-Button-Long-PressShort-Press/
> 
> Alternativamente, você pode usar https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d83/classAH_1_1IncrementButton.html , ele já lida com pressionamentos longos e curtos para você.

Obrigado tttapa, já consegui incluir o botão com duas funções usando a Control Surface

diegogauffin, 2024-05-29:
Hey Willame, can you share your code?