tttapa/Control-Surface #461: CCButtons<N> for potentiometers

TomW1605, 2021-04-26:
**Is your feature request related to a problem? Please describe.**
i am building a midi controller that is a mixer style for controlling voicemeeter. i am planning to have 7 channels all with a fader and a pan control. this means 14 potentiometers. this is a lot of the same declaration function over and over again.

**Describe the solution you'd like**
for buttons there is a solution: `CCButtons<N>`. this lets you define large numbers of buttons in a single declaration. i think the best solutions is this just for potentiometers

tttapa, 2021-05-13:
I'm on the fence about this, I don't think having 14 similar declarations is that problematic. You can just define them as (pin, address) pairs in an array as shown in this example:
https://github.com/tttapa/Control-Surface/blob/feee931d44d5f20fb7913b171895316e5819549d/examples/1.%20MIDI%20Output/1.%20Potentiometers%20%26%20Faders/Multiple-Control-Change-Potentiometers/Multiple-Control-Change-Potentiometers.ino#L44-L54

Being able to specify a range of pins, or a range of addresses might be handy in some cases, but increases the complexity and the number of classes that have to be maintained and documented. I'm open to the discussion, though.

TomW1605, 2021-05-14:
i some how missed that you could do them all as an array. i had put them all as individual declarations like this: https://pastebin.com/szrU2D3z. using the array will make it a lot simpler

closed 2021-05-14 by TomW1605